larixer/hid-asus-dkms #4: Freeze when touching touchpad

agrytnes, 2016-11-16:
After install I had to reboot a few times to be able to log in. It might have been that I touched the pad while loading Ubuntu. Now everything works fine until I touch the pad. When I do the PC freeze and caps look is blinking.

redmcg, 2016-11-17:
@agrytnes Before it freezes (and after it has booted) can you please send the output of:
`dmesg | grep hid`

You also might want to try the userspace driver (which can be found [here](https://github.com/redmcg/FTE1001)).

Make sure the hid_asus_fte module is not loaded when using the userspace driver. The hid_generic module should be loaded instead.

agrytnes, 2016-11-17:
This is output of dmesg | grep hid
[image]

[image]

vlasenko, 2016-11-17:
@agrytnes Please make sure this issue is happening on the latest driver source code.

https://github.com/vlasenko/hid-asus-fte-dkms#updating-to-the-latest-driver-version

agrytnes, 2016-11-17:
After latest driver source code ubuntu is no longer freezing and output looks different.

[image]

vlasenko, 2016-11-17:
Yes, the driver is not loaded now, however. I bet you don't have two finger scroll working now

agrytnes, 2016-11-17:
No two finger scrolling is correct. The only change is that the mouse cursor looks different and it changes to a pointing hand when over a link.

vlasenko, 2016-11-17:
Try the following sequence please:
1. Comment out MODULES_CONF line and REMAKE_INITRD line with prepending # in hid-asus-fte-dkms/dkms.conf
2. Run `./dkms-add.sh`
3. Add the following line to the end of `/etc/initramfs-tools/modules`:
```
hid_asus_fte
```
4. Update initrd:
```
sudo update-initramfs -u
```

vlasenko, 2016-11-17:
> No two finger scrolling is correct. The only change is that the mouse cursor looks different and it changes to a pointing hand when over a link.

I'm confused. According to the last your dmesg there is no `hid-focaltech` string in your logs, and only `hid-generic`. This means that the kernel is using it's stock driver for this touchpad. However, if you additionally run userspace driver from here: https://github.com/redmcg/FTE1001, then TouchPad will be handled by userspace driver. Please note that you should use either this DKMS driver or userspace one, not both. Please confirm that you are not using userspace driver and that your logs don't have `hid-focaltech` string and two finger scrolling works.

agrytnes, 2016-11-17:
Iv done 1. and 2. but im not sure what to do in 3. 

Im quite new to linux so im not sure where to look for FTE1001. I did clone it from github and i find the folder but i did not think I installed anything. How do I uninstall FTE1001 if i have it?

Thank you for your patience
Andreas

vlasenko, 2016-11-17:
@agrytnes I think you need to do 1-5 if you don't have two finger scroll working. If you do have it working, no need to do this. Please uncoment MODULES_CONF and REMAKE_INITRD lines again and run ./dkms-add

If you didn't install FTE1001 you would have known how to uninstall it, so no need to uninstall it. Simply cloning FTE1001 won't install anything.

vlasenko, 2016-11-17:
@agrytnes Let us know if TouchPad is working for you and we can close this issue

agrytnes, 2016-11-17:
Im working on step 4.

geting output:
[image]

redmcg, 2016-11-17:
They're just warnings and unrelated to this driver. You can ignore that. But that does mean you've completed step 4.

You can confirm the driver is in now your initrd by running:
`lsinitramfs /boot/initrd.img-`uname -r` | grep asus`

You should see output similar to:
`lib/modules/4.4.0-47-generic/updates/dkms/hid-asus-fte.ko`

If so the driver should now work after a reboot. If it isn't working - please run `dmesg | grep hid` again and provide the output.

agrytnes, 2016-11-17:
iv done step 1-5 and the touchpad is stil not working.

lsinitramfs /boot/initrd.img-uname -r | grep asus 
only gave:
/usr/bin/lsinitramfs: invalid option -- 'r'

dmesg | grep hid:
[image]

redmcg, 2016-11-17:
Sorry - that's my fault. I used a back tick in the command and Git uses that as a formatting character so it was removed. Run this instead:
`lsinitramfs /boot/initrd.img-$(uname -r) | grep asus`

The dmesg shows that the new driver is not being loaded. Can you try running:
`sudo modprobe hid_asus_fte`

and let me know the output?

agrytnes, 2016-11-17:
No worries! Im just happy to get help.

[image]

agrytnes, 2016-11-17:
[image]

redmcg, 2016-11-17:
OK - I think I know what the issue is now. It looks like you're using secure boot - and because our module isn't officially in the linux repos (i.e. it is considered third party) - it is being rejected.

You'll either need to disable secure boot or use the [userspace driver](https://github.com/redmcg/FTE1001).

The userspace driver it a program you run that talks to the device via useful interfaces provided by the kernel (`hidraw` and `uinput`). If you want to use the userspace driver I'll close this issue and we can open a new one in the [FTE1001 repo](https://github.com/redmcg/FTE1001) if required.

agrytnes, 2016-11-17:
I will try disable secure boot

agrytnes, 2016-11-17:
You were right! It works perfectly now! Big thanks to you and Victor!

closed 2016-11-17 by agrytnes