larixer/hid-asus-dkms #5: Naming

vlasenko, 2016-11-17:
This is an issue related to future inclusion of this source code to Linux kernel tree.

Right now in the source code it is mostly named FocalTech. I think it's not quite right. Because Vendor ID is ASUS. The firmware that implements Microsoft HID over I2C with vendor extensions has ASUS vendor id in HID descriptor. So I would call this TouchPad asus everywhere, or maybe asus_fte to not confuse it with devices that implement FocalTech I2C protocol, which is not HID protocol:
https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwiq-tjBha_QAhWHmBoKHWMBDlwQFggZMAA&url=http%3A%2F%2Fwww.newhavendisplay.com%2Fapp_notes%2FFT5x06_registers.pdf&usg=AFQjCNGfDLAi9Vg9oQBPUMxGMVaVMlnSgg&sig2=eqREkI2ERTF9GgI-R3LksQ&bvm=bv.139138859,d.d2s

Thoughts?

redmcg, 2016-11-17:
Agree 100%. I was just about to raise the same issue - so we're on exactly the same page.

I was going to go with the naming you are using i.e. 'asus-fte'. I'll change the driver name to be 'hid_asus_fte' and the function prefixes from 'focaltech_' to 'asus_'

vlasenko, 2016-11-17:
Yep, cool! 

Thank you!

redmcg, 2016-11-17:
I just made commit 7b5d66f28bad97239d5b49b5f1b60a28b3576a2d to address this.

vlasenko, 2016-11-17:
Looks good! Thanks a lot!

Victor

closed 2016-11-17 by vlasenko