larixer/hid-asus-dkms #6: Tap to click not working

omouren, 2016-11-17:
With Asus UX501VW, scrolling works great, good job!
But I lost the tap to click with this new driver.

vlasenko, 2016-11-17:
Hmm, this is strange, tap to click works for me as usual.

redmcg, 2016-11-17:
@omouren Can you please run:
`sudo evtest`

and then select the option that corresponds with the `FTE1001:00 0B05:0101` device.

This will dump every event sent by that device - which can be VERY verbose. Just try a quick and simple tap and provide the output.

If you need to install evtest - you can install it on Ubuntu via:
`sudo apt install evtest`

omouren, 2016-11-17:
Event: time 1479370989.502468, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 287
Event: time 1479370989.502468, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 1503
Event: time 1479370989.502468, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 720
Event: time 1479370989.502468, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 3
Event: time 1479370989.502468, type 3 (EV_ABS), code 58 (ABS_MT_PRESSURE), value 127
Event: time 1479370989.502468, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1479370989.502468, type 1 (EV_KEY), code 325 (BTN_TOOL_FINGER), value 1
Event: time 1479370989.502468, type 3 (EV_ABS), code 0 (ABS_X), value 1503
Event: time 1479370989.502468, type 3 (EV_ABS), code 1 (ABS_Y), value 720
Event: time 1479370989.502468, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 127
Event: time 1479370989.502468, -------------- SYN_REPORT ------------
Event: time 1479370989.604979, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 1
Event: time 1479370989.604979, type 3 (EV_ABS), code 58 (ABS_MT_PRESSURE), value 115
Event: time 1479370989.604979, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 115
Event: time 1479370989.604979, -------------- SYN_REPORT ------------
Event: time 1479370989.619029, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1479370989.619029, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1479370989.619029, type 1 (EV_KEY), code 325 (BTN_TOOL_FINGER), value 0
Event: time 1479370989.619029, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 0
Event: time 1479370989.619029, -------------- SYN_REPORT ------------

All events after a quick and simple tap like you said

redmcg, 2016-11-17:
Thanks - that's perfect. This shows the driver working correctly. Check in you system settings that 'Tap to click' is enabled.

It should show up now if you go to System Settings -> Mouse & Touchpad

omouren, 2016-11-17:
I'm on Linux Mint 18 (Kernel is 4.4.0-47-generic), I already enabled all of these features in my system settings but I think that it didn't change anything...

redmcg, 2016-11-17:
Try running:
`synclient | grep 'Tap\|Finger\(High\|Low\)\|ClickTime'`

You can then compare your values to mine:
```
$ synclient | grep 'Tap\|Finger\(High\|Low\)\|ClickTime'
    FingerLow               = 12
    FingerHigh              = 15
    MaxTapTime              = 180
    MaxTapMove              = 145
    MaxDoubleTapTime        = 100
    SingleTapTimeout        = 180
    ClickTime               = 100
    TapButton1              = 1
    TapButton2              = 3
    TapButton3              = 0
    TapAndDragGesture       = 1
```

I'm not super familiar with synaptics - but the `man` page looks comprehensive. Run `man synaptics` to view it.

An extract from the Tapping section is:

> A tap event happens when the finger is touched and released in a time interval shorter than MaxTapTime, and the touch and release coordinates are less than MaxTapMove units apart.  A "touch" event happens when the Z value goes above FingerHigh, and an "untouch" event happens when the Z value goes below FingerLow.

Z refers the value of `ABS_PRESSURE` from the log you provided earlier. So Z was `127` at 1479370989.502468 and `0` at 1479370989.619029. That means your tap lasted 116.561 msecs. Which is less than my value of `180` for `MaxTapTime`. So on my machine - your tap would have registered.

omouren, 2016-11-17:
synclient doesn't work :
Couldn't find synaptics properties. No synaptics driver loaded?

redmcg, 2016-11-17:
What does `xinput list-props "FTE1001:00 0B05:0101"` show?

omouren, 2016-11-17:
Device 'FTE1001:00 0B05:0101':
	Device Enabled (168):	1
	Coordinate Transformation Matrix (170):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	libinput Tapping Enabled (304):	0
	libinput Tapping Enabled Default (305):	0
	libinput Tapping Drag Enabled (306):	1
	libinput Tapping Drag Enabled Default (307):	1
	libinput Tapping Drag Lock Enabled (308):	0
	libinput Tapping Drag Lock Enabled Default (309):	0
	libinput Accel Speed (310):	0.000000
	libinput Accel Speed Default (311):	0.000000
	libinput Natural Scrolling Enabled (312):	0
	libinput Natural Scrolling Enabled Default (313):	0
	libinput Send Events Modes Available (288):	1, 1
	libinput Send Events Mode Enabled (289):	0, 0
	libinput Send Events Mode Enabled Default (290):	0, 0
	libinput Left Handed Enabled (314):	0
	libinput Left Handed Enabled Default (315):	0
	libinput Scroll Methods Available (316):	1, 1, 0
	libinput Scroll Method Enabled (317):	1, 0, 0
	libinput Scroll Method Enabled Default (318):	1, 0, 0
	libinput Click Methods Available (319):	1, 1
	libinput Click Method Enabled (320):	1, 0
	libinput Click Method Enabled Default (321):	1, 0
	libinput Disable While Typing Enabled (322):	1
	libinput Disable While Typing Enabled Default (323):	1
	Device Node (291):	"/dev/input/event12"
	Device Product ID (292):	2821, 257
	libinput Drag Lock Buttons (324):	<no items>
	libinput Horizonal Scroll Enabled (293):	1

redmcg, 2016-11-17:
OK - looks like you're using libinput. I've never used it myself so from here on in I'm guessing. But I would try:
```
xinput set-prop "FTE1001:00 0B05:0101" "libinput Tapping Enabled" 1
xinput set-prop "FTE1001:00 0B05:0101" "libinput Tapping Enabled Default" 1
```

omouren, 2016-11-17:
It works after the first command thanks. But it still strange that Mint can't change touchpad settings.

redmcg, 2016-11-17:
Agreed. It is strange. But glad to hear it's working. Enjoy the tapping and the scrolling!

closed 2016-11-17 by redmcg

omouren, 2016-11-17:
Fixed Mint settings after remove xserver-xorg-input-libinput package.

ceejayrao, 2016-11-18:
On my system libinput is the driver, I initially also had the same issue of pointer not working for tap, then after configuring the touchpad settings to enable tap click, every thing started to work.

I am running kubuntu 16.04 with kernel 4.8.3. The only issue I see is that all options to configure the touchpad are not available for change. I need to see how to get that to work or will a different issue when I have gathered all data.

Thanks
Jitendra.