larixer/hid-asus-dkms #7: Driver Does Not Work

kah0922, 2016-11-17:
I have an ASUS X540LA running the lastest openSUSE Tumbleweed. I installed the driver, but my touchpad is still recognized as a mouse.

kah0922, 2016-11-17:
Wait, never mind. Didn't blacklist hic_generic. It's working now.

closed 2016-11-17 by kah0922

redmcg, 2016-11-17:
Glad to hear it's working. But you shouldn't need to blacklist `hid_generic` (and you might need it if you plug in a USB mouse for example). The new driver should be picked up if it is in your initrd.

I'm not 100% sure how to add it to initrd in openSUSE - but for Ubuntu you need to edit `/etc/initramfs-tools/modules` and add a single entry of:
`hid_asus_fte`

And then rebuild your initrd with:
`update-initramfs -u`

It's up to you if you want to try this - but it would be good to have confirmation that this driver works in openSUSE without having to blacklist hid_generic.

redmcg, 2016-11-17:
I should also point out that you can check the driver is in your initrd by running:
`lsinitramfs /boot/initrd.img-$(uname -r) | grep asus`