larixer/hid-asus-dkms #10: FTE Touchpad not found instead of executing the driver

carmilso, 2016-11-18:
When I execute the dkms-add.sh I receive the following output:

`hid-asus-fte-dkms (master|✔) ./dkms-add.sh                        

-------- Uninstall Beginning --------
Module:  asus-fte
Version: 1.0
Kernel:  4.8.8-2-ARCH (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

hid-asus-fte.ko:
 - Uninstallation
   - Deleting from: /usr/lib/modules/4.8.8-2-ARCH/updates/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.

------------------------------
Deleting module version: 1.0
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/asus-fte/1.0/source ->
                 /usr/src/asus-fte-1.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=4.8.8-2-ARCH -C /usr/lib/modules/4.8.8-2-ARCH/build M=/var/lib/dkms/asus-fte/1.0/build/src hid-asus-fte.ko...
cleaning build area...
Kernel cleanup unnecessary for this kernel.  Skipping...

DKMS: build completed.

hid-asus-fte.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/4.8.8-2-ARCH/updates/

depmod...

DKMS: install completed.`

But multi-gesture on touchpad doesn't work.

And If I execute xinput list I receive:

`hid-asus-fte-dkms (master|✔) xinput list      
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ Logitech M325                             id=12   [slave  pointer  (2)]
⎜   ↳ FTE1001:00 0B05:0101                      id=15   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Asus Wireless Radio Control               id=7    [slave  keyboard (3)]
    ↳ Video Bus                                 id=8    [slave  keyboard (3)]
    ↳ Video Bus                                 id=9    [slave  keyboard (3)]
    ↳ Sleep Button                              id=10   [slave  keyboard (3)]
    ↳ USB2.0 HD UVC WebCam                      id=11   [slave  keyboard (3)]
    ↳ Asus WMI hotkeys                          id=13   [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard              id=14   [slave  keyboard (3)]
`
But no touchpad is found

carmilso, 2016-11-18:
System is ArchLinux 4.8.8-2 with KDE and the laptop is Asus Zenbook Pro (UX501)

vlasenko, 2016-11-18:
Hi,

This mean the driver is not loaded into memory.

What is the output of 
`lsmod | grep hid`
?

carmilso, 2016-11-18:
`hid-asus-fte-dkms (master|✔) lsmod | grep hid
hid_asus_fte           16384  0
i2c_hid                20480  0
hid_generic            16384  0
hid_logitech_hidpp     28672  0
hid_logitech_dj        20480  0
usbhid                 49152  0
mac_hid                16384  0
hid                   114688  6 hid_asus_fte,i2c_hid,hid_generic,usbhid,hid_logitech_dj,hid_logitech_hidpp
usbcore               208896  5 uvcvideo,usbhid,xhci_pci,btusb,xhci_hcd
`

vlasenko, 2016-11-18:
Try to remove `hid_generic`
`sudo modprobe -r hid_generic`

Does it really removes it?
`sudo lsmod | grep hid`

carmilso, 2016-11-18:
This is the new output after removing the hid_generic:

`
hid_asus_fte           16384  0
i2c_hid                20480  0
hid_logitech_hidpp     28672  0
hid_logitech_dj        20480  0
usbhid                 49152  0
mac_hid                16384  0
hid                   114688  5 hid_asus_fte,i2c_hid,usbhid,hid_logitech_dj,hid_logitech_hidpp
usbcore               208896  5 uvcvideo,usbhid,xhci_pci,btusb,xhci_hcd
`

But I can't now see the mouse

vlasenko, 2016-11-18:
`sudo modprobe -r hid_asus_fte`
`sudo modprobe hid_asus_fte`
and paste please
`tail -10 dmesg`

carmilso, 2016-11-18:
Not found the file dmesg

vlasenko, 2016-11-18:
Sorry
`dmesg | tail -10`

carmilso, 2016-11-18:
Sorry, it's true!

[  408.602412] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-8/i2c-FTE1001:00/0018:0B05:0101.0006/input/input17

[  408.602820] hid-asus-fte 0018:0B05:0101.0006: input,hidraw2: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00

[  428.983628] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:633 / ret_size:0

[  428.987027] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:131 / ret_size:0

[  428.987171] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-8/i2c-FTE1001:00/0018:0B05:0101.0007/input/input18

[  428.987572] hid-generic 0018:0B05:0101.0007: input,hidraw2: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00

[ 1916.324955] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:633 / ret_size:0

[ 1916.328353] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:131 / ret_size:0

[ 1916.328498] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-8/i2c-FTE1001:00/0018:0B05:0101.0008/input/input19

[ 1916.328867] hid-generic 0018:0B05:0

vlasenko, 2016-11-18:
Okay, seems the driver was loaded once (the line hid-asus-fte), but then, the control of the device has been taken over by `hid-generic`.

Let's try to gain control of the TouchPad on boot. You need to add hid_asus_fte module to initrd.

I'm not sure how to do it on Arch - but for Ubuntu you need to edit /etc/initramfs-tools/modules and add a single entry of:
hid_asus_fte

And then rebuild your initrd with:
update-initramfs -u

And reboot

vlasenko, 2016-11-18:
To test that driver is working when loaded you can execute
`./dev-run.sh` from driver source folder, it should build the driver and load it into memory temporary, until reboot.

carmilso, 2016-11-18:
Following https://bbs.archlinux.org/viewtopic.php?id=209554&p=2 I have added hid_asus_fte to /etc/modules-load.d/touchpad.conf but when I reboot the laptop it does not show the mouse

vlasenko, 2016-11-18:
`dmesg | grep hid`
?

carmilso, 2016-11-18:
[    2.084141] hidraw: raw HID events driver (C) Jiri Kosina
[    2.086519] hid_asus_fte: loading out-of-tree module taints kernel.
[    2.579429] usbcore: registered new interface driver usbhid
[    2.579430] usbhid: USB HID core driver
[    2.594385] logitech-djreceiver 0003:046D:C52B.0003: hiddev0,hidraw0: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:00:14.0-6/input2
[    2.721996] logitech-hidpp-device 0003:046D:400A.0004: input,hidraw1: USB HID v1.11 Mouse [Logitech M325] on usb-0000:00:14.0-6:1
[    7.750632] i2c_hid i2c-FTE1001:00: failed to reset device.
[   13.937320] i2c_hid i2c-FTE1001:00: failed to reset device.
[   14.956065] hid-asus-fte 0018:0B05:0101.0005: input,hidraw2: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00

vlasenko, 2016-11-18:
Does executing `./dev-run.sh` from source code folder activates mouse for you?

carmilso, 2016-11-18:
I can only reactivate the mouse executing dev-restore.sh

vlasenko, 2016-11-18:
Does the line `i2c_hid i2c-FTE1001:00: failed to reset device` appears each time you run `./dev-run.sh` in the dmesg log ? I assume it is the cause of the problem

carmilso, 2016-11-18:
If I don't execute the dev-run.sh I have the following output:

[    2.021311] hidraw: raw HID events driver (C) Jiri Kosina
[    2.024023] hid_asus_fte: loading out-of-tree module taints kernel.
[    2.491963] usbcore: registered new interface driver usbhid
[    2.491964] usbhid: USB HID core driver
[    2.496627] logitech-djreceiver 0003:046D:C52B.0003: hiddev0,hidraw0: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:00:14.0-6/input2
[    2.623074] logitech-hidpp-device 0003:046D:400A.0004: input,hidraw1: USB HID v1.11 Mouse [Logitech M325] on usb-0000:00:14.0-6:1
[    7.742502] i2c_hid i2c-FTE1001:00: failed to reset device.
[   13.929127] i2c_hid i2c-FTE1001:00: failed to reset device.

vlasenko, 2016-11-18:
Please try to update source code now
`git pull`
, run 
`./dev-run.sh`
and report back whether mouse appears and scroll works.

carmilso, 2016-11-18:
Okey! When I reboot now it works perfectly without executing dev-run.sh and if I execute it it works well the same.
This is the output now for dmesg|grep hid:

[    2.084601] hidraw: raw HID events driver (C) Jiri Kosina
[    2.090926] hid_asus_fte: loading out-of-tree module taints kernel.
[    2.718458] usbcore: registered new interface driver usbhid
[    2.718459] usbhid: USB HID core driver
[    2.725766] logitech-djreceiver 0003:046D:C52B.0003: hiddev0,hidraw0: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:00:14.0-6/input2
[    2.847751] hid-asus-fte 0018:0B05:0101.0005: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
[    2.852892] logitech-hidpp-device 0003:046D:400A.0004: input,hidraw2: USB HID v1.11 Mouse [Logitech M325] on usb-0000:00:14.0-6:1
[  148.408865] hid-asus-fte 0018:0B05:0101.0006: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00

And this is the output for xinput list:

⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ Logitech M325                             id=12   [slave  pointer  (2)]
⎜   ↳ Asus FTE TouchPad                         id=13   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Asus Wireless Radio Control               id=7    [slave  keyboard (3)]
    ↳ Video Bus                                 id=8    [slave  keyboard (3)]
    ↳ Video Bus                                 id=9    [slave  keyboard (3)]
    ↳ Sleep Button                              id=10   [slave  keyboard (3)]
    ↳ USB2.0 HD UVC WebCam                      id=11   [slave  keyboard (3)]
    ↳ Asus WMI hotkeys                          id=14   [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard              id=15   [slave  keyboard (3)]

vlasenko, 2016-11-18:
Fine, thanks. I'm closing the issue for now.

closed 2016-11-18 by vlasenko

carmilso, 2016-11-18:
Okey, thanks!