larixer/hid-asus-dkms #8: Loaded but not working

mdhsl, 2016-11-18:
From bugzilla:
It still does not work for me. I am thinking about a conflict between modules. Some of you have an idea about how to fix that and making this file working for me?
Here my lsmod: 

lsmod 
Module                  Size  Used by
rfcomm                 55936  12
bnep                   13597  2
hid_generic             1385  0
hid_asus_fte            3319  0
btusb                  32441  0
btrtl                   4832  1 btusb
i2c_designware_platform     4674  0
i2c_designware_core     8582  1 i2c_designware_platform
asus_nb_wmi            13264  0
asus_wmi               17456  1 asus_nb_wmi
mxm_wmi                 1571  0
iwlmvm                252163  0
nvidia_drm             36977  0
nvidia_modeset        754552  1 nvidia_drm
nvidia              11800821  1 nvidia_modeset
iwlwifi               176315  1 iwlmvm
x86_pkg_temp_thermal     7735  0
serio_raw               5041  0
i2c_i801               18922  0
i2c_smbus               3617  1 i2c_i801
intel_lpss_pci          4622  0
processor_thermal_device     6122  0
intel_soc_dts_iosf      4926  1 processor_thermal_device
elan_i2c               22413  0
i2c_hid                11884  0
int3403_thermal         2536  0
wmi                     6996  2 asus_wmi,mxm_wmi
pinctrl_sunrisepoint    15059  0
pinctrl_intel          10374  1 pinctrl_sunrisepoint
intel_lpss_acpi         2449  0
int3402_thermal         1720  0
intel_lpss              4829  2 intel_lpss_pci,intel_lpss_acpi
int3400_thermal         4070  0
int340x_thermal_zone     3204  3 int3402_thermal,int3403_thermal,processor_thermal_device
acpi_thermal_rel        4330  1 int3400_thermal
efivarfs                5351  1

I don't use dkms because I compiles my own kernel( Gentoo dist).
I've followed the steps defined into the github:
1) rmmod hid_asus_fte i2c_hid hid_generic
2) insmod src/hid-asus-fte.ko 
3) modprobe i2c_hid
------------------------------
@nvsl I can see you are missing the 'hid' module. So try running modprobe 'hid' between your steps 1 and 2. But if you continue to have issues please raise an issue at the GitHub repository:
https://github.com/vlasenko/hid-asus-fte-dkms
-------------------------------

=> Into the make menuconfig, I have only hid_generic or uhid. The hid module is forced to "y".
I tried a lot of things but it still does not work. The strange thing is if I load the i2c_designware module, the touchpad is reacting. I can put my two fingers on it and the cursor will switch over the two locations continuously.  The nodejs example (the one to switch the touchpad into the touchpad) is working great by the way. So I don't know what's wrong with my configuration :/

redmcg, 2016-11-18:
Can you please try the following:
```
modprobe -r hid_asus_fte hid_generic i2c_hid
insmod src/hid-asus-fte.ko
modprobe i2c_hid
```
and then provide the output of `dmesg | grep hid`

mdhsl, 2016-11-18:
dmesg | grep hid
[    0.922631] hidraw: raw HID events driver (C) Jiri Kosina
[    0.924181] usbcore: registered new interface driver usbhid
[    0.924215] usbhid: USB HID core driver
[    1.799973] logitech-djreceiver 0003:046D:C52B.0003: hiddev0,hidraw0: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:00:14.0-6/input2
[    1.912385] logitech-hidpp-device 0003:046D:101B.0004: input,hidraw1: USB HID v1.11 Mouse [Logitech M705] on usb-0000:00:14.0-6:1
[  123.537277] uhid: unknown parameter 'hid_generic' ignored

--------------------
masamune linux # lsmod 
Module                  Size  Used by
i2c_hid                11884  0
hid_asus_fte            3319  0
rfcomm                 55936  12
bnep                   13597  2
asus_nb_wmi            13264  0
asus_wmi               17456  1 asus_nb_wmi
mxm_wmi                 1571  0
btusb                  32441  0
btrtl                   4832  1 btusb
iwlmvm                252163  0
nvidia_drm             36977  0
nvidia_modeset        754552  1 nvidia_drm
nvidia              11800821  1 nvidia_modeset
iwlwifi               176315  1 iwlmvm
x86_pkg_temp_thermal     7735  0
serio_raw               5041  0
i2c_i801               18922  0
i2c_smbus               3617  1 i2c_i801
intel_lpss_pci          4622  0
processor_thermal_device     6122  0
intel_soc_dts_iosf      4926  1 processor_thermal_device
int3403_thermal         2536  0
wmi                     6996  2 asus_wmi,mxm_wmi
pinctrl_sunrisepoint    15059  0
pinctrl_intel          10374  1 pinctrl_sunrisepoint
intel_lpss_acpi         2449  0
int3402_thermal         1720  0
intel_lpss              4829  2 intel_lpss_pci,intel_lpss_acpi
int340x_thermal_zone     3204  3 int3402_thermal,int3403_thermal,processor_thermal_device
int3400_thermal         4070  0
acpi_thermal_rel        4330  1 int3400_thermal
efivarfs                5351  1

redmcg, 2016-11-18:
Try `modprobe i2c_designware_platform` and send me the output of `dmesg | grep hid` again please

mdhsl, 2016-11-18:
Here the output:

masamune linux # modprobe i2c_designware_platform
masamune linux # dmesg | grep hid
[    0.923000] hidraw: raw HID events driver (C) Jiri Kosina
[    0.924573] usbcore: registered new interface driver usbhid
[    0.924576] usbhid: USB HID core driver
[    1.798230] logitech-djreceiver 0003:046D:C52B.0003: hiddev0,hidraw0: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:00:14.0-6/input2
[    1.909644] logitech-hidpp-device 0003:046D:101B.0004: input,hidraw1: USB HID v1.11 Mouse [Logitech M705] on usb-0000:00:14.0-6:1
[  239.288752] hid-focaltech 0018:0B05:0101.0005: input,hidraw2: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00

mdhsl, 2016-11-18:
I've recorded my screen for a few seconds, Maybe it will help to understand what's happen:

http://s000.tinyupload.com/?file_id=75873596885469090653

When I click on a location on the touchPad, the cursor is moving directly to that position

redmcg, 2016-11-18:
OK - so the good news is - this line:
`[ 239.288752] hid-focaltech 0018:0B05:0101.0005: input,hidraw2: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00`
shows that driver is loaded and that the device has been bound to it.

Can you please run:
`evtest`

and then select the option that corresponds with the `Asus FTE TouchPad` device (it was called `FTE1001:00 0B05:0101` in the older versions of the driver).

It's very verbose - so please just do a very quick test (a single tap for example).

mdhsl, 2016-11-18:
Thanks for helping me, here the output:

evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Sleep Button
/dev/input/event10:	HDA Intel PCH HDMI/DP
/dev/input/event11:	USB2.0 HD UVC WebCam
/dev/input/event12:	Logitech M705
/dev/input/event13:	Asus WMI hotkeys
/dev/input/event14:	Asus FTE TouchPad
/dev/input/event2:	Power Button
/dev/input/event3:	Video Bus
/dev/input/event4:	Video Bus
/dev/input/event5:	PC Speaker
/dev/input/event6:	AT Translated Set 2 keyboard
/dev/input/event7:	HDA Intel PCH Headphone
/dev/input/event8:	HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event9:	HDA Intel PCH HDMI/DP,pcm=7
Select the device event number [0-14]: 14
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0xb05 product 0x101 version 0x100
Input device name: "Asus FTE TouchPad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 325 (BTN_TOOL_FINGER)
    Event code 330 (BTN_TOUCH)
    Event code 333 (BTN_TOOL_DOUBLETAP)
    Event code 334 (BTN_TOOL_TRIPLETAP)
    Event code 335 (BTN_TOOL_QUADTAP)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value   1051
      Min        0
      Max     2794
    Event code 1 (ABS_Y)
      Value    997
      Min        0
      Max     1758
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max      128
    Event code 28 (ABS_TOOL_WIDTH)
      Value      0
      Min        0
      Max        8
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        4
    Event code 48 (ABS_MT_TOUCH_MAJOR)
      Value      0
      Min        0
      Max        8
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max     2794
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max     1758
    Event code 55 (ABS_MT_TOOL_TYPE)
      Value      0
      Min        0
      Max        2
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
    Event code 58 (ABS_MT_PRESSURE)
      Value      0
      Min        0
      Max      128
Properties:
  Property type 0 (INPUT_PROP_POINTER)
  Property type 2 (INPUT_PROP_BUTTONPAD)
Testing ... (interrupt to exit)
Event: time 1479439183.240250, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 2
Event: time 1479439183.240250, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 1293
Event: time 1479439183.240250, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 683
Event: time 1479439183.240250, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 3
Event: time 1479439183.240250, type 3 (EV_ABS), code 58 (ABS_MT_PRESSURE), value 127
Event: time 1479439183.240250, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1479439183.240250, type 1 (EV_KEY), code 325 (BTN_TOOL_FINGER), value 1
Event: time 1479439183.240250, type 3 (EV_ABS), code 0 (ABS_X), value 1293
Event: time 1479439183.240250, type 3 (EV_ABS), code 1 (ABS_Y), value 683
Event: time 1479439183.240250, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 127
Event: time 1479439183.240250, -------------- SYN_REPORT ------------
Event: time 1479439183.333069, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 1292
Event: time 1479439183.333069, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 687
Event: time 1479439183.333069, type 3 (EV_ABS), code 0 (ABS_X), value 1292
Event: time 1479439183.333069, type 3 (EV_ABS), code 1 (ABS_Y), value 687
Event: time 1479439183.333069, -------------- SYN_REPORT ------------
Event: time 1479439183.393282, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 1
Event: time 1479439183.393282, type 3 (EV_ABS), code 58 (ABS_MT_PRESSURE), value 92
Event: time 1479439183.393282, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 92
Event: time 1479439183.393282, -------------- SYN_REPORT ------------
Event: time 1479439183.407781, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1479439183.407781, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1479439183.407781, type 1 (EV_KEY), code 325 (BTN_TOOL_FINGER), value 0
Event: time 1479439183.407781, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 0
Event: time 1479439183.407781, -------------- SYN_REPORT ------------

mdhsl, 2016-11-18:
xinput
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Logitech M705                           	id=11	[slave  pointer  (2)]
⎜   ↳ Asus FTE TouchPad                       	id=14	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ USB2.0 HD UVC WebCam                    	id=10	[slave  keyboard (3)]
    ↳ Asus WMI hotkeys                        	id=12	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=13	[slave  keyboard (3)]

------------------------
linux # xinput list-props "Asus FTE TouchPad"
Device 'Asus FTE TouchPad':
	Device Enabled (138):	1
	Coordinate Transformation Matrix (140):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (270):	0
	Device Accel Constant Deceleration (271):	1.000000
	Device Accel Adaptive Deceleration (272):	1.000000
	Device Accel Velocity Scaling (273):	10.000000
	Device Product ID (262):	2821, 257
	Device Node (261):	"/dev/input/event14"
	Evdev Axis Inversion (598):	0, 0
	Evdev Axis Calibration (599):	<no items>
	Evdev Axes Swap (600):	0
	Axis Labels (601):	"Abs MT Position X" (576), "Abs MT Position Y" (577), "Abs MT Pressure" (575), "Abs Tool Width" (596), "Abs MT Touch Major" (572), "Abs MT Tool Type" (597), "None" (0), "None" (0), "None" (0)
	Button Labels (602):	"Button Left" (141), "Button Unknown" (595), "Button Unknown" (595), "Button Wheel Up" (144), "Button Wheel Down" (145)
	Evdev Scrolling Distance (603):	0, 0, 0
	Evdev Middle Button Emulation (604):	0
	Evdev Middle Button Timeout (605):	50
	Evdev Third Button Emulation (606):	0
	Evdev Third Button Emulation Timeout (607):	1000
	Evdev Third Button Emulation Button (608):	3
	Evdev Third Button Emulation Threshold (609):	20
	Evdev Wheel Emulation (610):	0
	Evdev Wheel Emulation Axes (611):	0, 0, 4, 5
	Evdev Wheel Emulation Inertia (612):	10
	Evdev Wheel Emulation Timeout (613):	200
	Evdev Wheel Emulation Button (614):	4
	Evdev Drag Lock Buttons (615):	0
----------------------------

The thing is for the evtest, I only made a single tap (with 1 finger). In the video, I make a tap using two fingers and keep maintains over the time. If I reproduced what we see into the video using evtest, it gives that:
....
Event: time 1479439827.521770, -------------- SYN_REPORT ------------
Event: time 1479439827.559695, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 777
Event: time 1479439827.559695, type 3 (EV_ABS), code 1 (ABS_Y), value 777
Event: time 1479439827.559695, -------------- SYN_REPORT ------------
Event: time 1479439827.567175, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 778
Event: time 1479439827.567175, type 3 (EV_ABS), code 1 (ABS_Y), value 778
Event: time 1479439827.567175, -------------- SYN_REPORT ------------
Event: time 1479439827.574600, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 779
Event: time 1479439827.574600, type 3 (EV_ABS), code 47 (ABS_MT_SLOT), value 1
Event: time 1479439827.574600, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 549
Event: time 1479439827.574600, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 4
Event: time 1479439827.574600, type 3 (EV_ABS), code 1 (ABS_Y), value 779
Event: time 1479439827.574600, -------------- SYN_REPORT ------------
Event: time 1479439827.582098, type 3 (EV_ABS), code 47 (ABS_MT_SLOT), value 0
Event: time 1479439827.582098, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 778
Event: time 1479439827.582098, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 2
Event: time 1479439827.582098, type 3 (EV_ABS), code 47 (ABS_MT_SLOT), value 1
Event: time 1479439827.582098, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 3
Event: time 1479439827.582098, type 3 (EV_ABS), code 1 (ABS_Y), value 778
Event: time 1479439827.582098, -------------- SYN_REPORT ------------
Event: time 1479439827.590000, type 3 (EV_ABS), code 47 (ABS_MT_SLOT), value 0
Event: time 1479439827.590000, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 3
[... 57 lines cut ...]
Event: time 1479439827.830302, type 1 (EV_KEY), code 325 (BTN_TOOL_FINGER), value 1
Event: time 1479439827.830302, type 1 (EV_KEY), code 333 (BTN_TOOL_DOUBLETAP), value 0
Event: time 1479439827.830302, type 3 (EV_ABS), code 0 (ABS_X), value 1879
Event: time 1479439827.830302, type 3 (EV_ABS), code 1 (ABS_Y), value 549
Event: time 1479439827.830302, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 109
Event: time 1479439827.830302, -------------- SYN_REPORT ------------
Event: time 1479439827.837875, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 1878
Event: time 1479439827.837875, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 552
Event: time 1479439827.837875, type 3 (EV_ABS), code 58 (ABS_MT_PRESSURE), value 76
Event: time 1479439827.837875, type 3 (EV_ABS), code 0 (ABS_X), value 1878
Event: time 1479439827.837875, type 3 (EV_ABS), code 1 (ABS_Y), value 552
Event: time 1479439827.837875, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 76
[... cut ...]
Event: time 1479439827.845051, type 3 (EV_ABS), code 1 (ABS_Y), value 556
Event: time 1479439827.845051, -------------- SYN_REPORT ------------
Event: time 1479439827.852283, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1479439827.852283, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1479439827.852283, type 1 (EV_KEY), code 325 (BTN_TOOL_FINGER), value 0
Event: time 1479439827.852283, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 0
Event: time 1479439827.852283, -------------- SYN_REPORT ------------

....

mdhsl, 2016-11-18:
Ok Guys, I've figure out what's going on. I've removed xf86-input-synaptics and it works!!! touchpad and scroll.

Thank you very much guys for you work on this driver and your support!

closed 2016-11-18 by mdhsl

redmcg, 2016-11-18:
Good work! Thanks Mathieu

mdhsl, 2016-11-18:
Is the double click is handled by the driver? (by double tapping)

redmcg, 2016-11-18:
The driver is only responsible for reporting the individual taps. It'll be your Xorg input driver for Touchpads (currently Evdev) that is responsible for recognising and interpreting it (i.e. performing a configured action). I'm not 100% familiar with Evdev's capabilities though. I do know you need to use Synaptics for some features.

mdhsl, 2016-11-18:
Perfect! I'll find the right Xorg drivers to make the tap/click working. Thanks for precisions