larixer/hid-asus-dkms #9: Pointer stops responding

chrisrobinson76, 2016-11-18:
I've tested on my Asus X540S running Linux Mint 17.3 with KDE, kernel 4.4.0-47-generic with a build from 23.00 GMT on 17th November. What I've found is that the touchpad works fine from boot for a while and the multitouch features work as expected. At some point though, the pointer stops responding/moving although I can still scroll a page up and down using the touchpad and use the buttons to click. If I then switch to a console (e.g. ctrl-alt-f7) and then back (ctrl-alt-f8) the touchpad functionality is fully restored until the next time it stops responding.
Thanks for looking into this.

vlasenko, 2016-11-18:
@chrisrobinson76  When TouchPad is not responding, could you please run:
`sudo evtest`

and then select the option that corresponds to the `Asus FTE TouchPad` device. After that try to move the cursor and copy paste evtest output here.

If you need to install evtest - you can install it on Ubuntu via:
`sudo apt install evtest`

vlasenko, 2016-11-18:
Also please see this issue:
https://github.com/vlasenko/hid-asus-fte-dkms/issues/6

Perhaps you need to remove libinput as well?

chrisrobinson76, 2016-11-18:
I'll try removing libinput now. here's the evtest output:
Event: time 1479459235.961009, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1509
Event: time 1479459235.961009, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 3
Event: time 1479459235.961009, type 3 (EV_ABS), code 0 (ABS_X), value 887
Event: time 1479459235.961009, type 3 (EV_ABS), code 1 (ABS_Y), value 1509
Event: time 1479459235.961009, -------------- SYN_REPORT ------------
Event: time 1479459235.968703, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 2
Event: time 1479459235.968703, -------------- SYN_REPORT ------------
Event: time 1479459235.976442, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 3
Event: time 1479459235.976442, -------------- SYN_REPORT ------------
Event: time 1479459235.984066, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 885
Event: time 1479459235.984066, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1513
Event: time 1479459235.984066, type 3 (EV_ABS), code 0 (ABS_X), value 885
Event: time 1479459235.984066, type 3 (EV_ABS), code 1 (ABS_Y), value 1513
Event: time 1479459235.984066, -------------- SYN_REPORT ------------
Event: time 1479459235.998317, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 883
Event: time 1479459235.998317, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1517
Event: time 1479459235.998317, type 3 (EV_ABS), code 0 (ABS_X), value 883
Event: time 1479459235.998317, type 3 (EV_ABS), code 1 (ABS_Y), value 1517
Event: time 1479459235.998317, -------------- SYN_REPORT ------------
Event: time 1479459236.005882, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 882
Event: time 1479459236.005882, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1519
Event: time 1479459236.005882, type 3 (EV_ABS), code 0 (ABS_X), value 882
Event: time 1479459236.005882, type 3 (EV_ABS), code 1 (ABS_Y), value 1519
Event: time 1479459236.005882, -------------- SYN_REPORT ------------
Event: time 1479459236.013719, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 881
Event: time 1479459236.013719, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1521
Event: time 1479459236.013719, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 2
Event: time 1479459236.013719, type 3 (EV_ABS), code 0 (ABS_X), value 881
Event: time 1479459236.013719, type 3 (EV_ABS), code 1 (ABS_Y), value 1521
Event: time 1479459236.013719, -------------- SYN_REPORT ------------
Event: time 1479459236.021254, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 880
Event: time 1479459236.021254, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1524
Event: time 1479459236.021254, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 3
Event: time 1479459236.021254, type 3 (EV_ABS), code 0 (ABS_X), value 880
Event: time 1479459236.021254, type 3 (EV_ABS), code 1 (ABS_Y), value 1524
Event: time 1479459236.021254, -------------- SYN_REPORT ------------
Event: time 1479459236.028328, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 879
Event: time 1479459236.028328, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1527
Event: time 1479459236.028328, type 3 (EV_ABS), code 0 (ABS_X), value 879
Event: time 1479459236.028328, type 3 (EV_ABS), code 1 (ABS_Y), value 1527
Event: time 1479459236.028328, -------------- SYN_REPORT ------------
Event: time 1479459236.035635, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 878
Event: time 1479459236.035635, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1529
Event: time 1479459236.035635, type 3 (EV_ABS), code 0 (ABS_X), value 878
Event: time 1479459236.035635, type 3 (EV_ABS), code 1 (ABS_Y), value 1529
Event: time 1479459236.035635, -------------- SYN_REPORT ------------
Event: time 1479459236.043336, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1531
Event: time 1479459236.043336, type 3 (EV_ABS), code 1 (ABS_Y), value 1531
Event: time 1479459236.043336, -------------- SYN_REPORT ------------
Event: time 1479459236.050601, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1533
Event: time 1479459236.050601, type 3 (EV_ABS), code 1 (ABS_Y), value 1533
Event: time 1479459236.050601, -------------- SYN_REPORT ------------
Event: time 1479459236.057380, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 877
Event: time 1479459236.057380, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1535
Event: time 1479459236.057380, type 3 (EV_ABS), code 0 (ABS_X), value 877
Event: time 1479459236.057380, type 3 (EV_ABS), code 1 (ABS_Y), value 1535
Event: time 1479459236.057380, -------------- SYN_REPORT ------------
Event: time 1479459236.065213, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1536
Event: time 1479459236.065213, type 3 (EV_ABS), code 1 (ABS_Y), value 1536
Event: time 1479459236.065213, -------------- SYN_REPORT ------------
Event: time 1479459236.072992, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1537
Event: time 1479459236.072992, type 3 (EV_ABS), code 1 (ABS_Y), value 1537
Event: time 1479459236.072992, -------------- SYN_REPORT ------------
Event: time 1479459236.080355, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1538
Event: time 1479459236.080355, type 3 (EV_ABS), code 1 (ABS_Y), value 1538
Event: time 1479459236.080355, -------------- SYN_REPORT ------------
Event: time 1479459236.095145, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1539
Event: time 1479459236.095145, type 3 (EV_ABS), code 1 (ABS_Y), value 1539
Event: time 1479459236.095145, -------------- SYN_REPORT ------------
[... cut ...]
Event: time 1479459239.819822, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 49
Event: time 1479459239.819822, -------------- SYN_REPORT ------------
Event: time 1479459239.833358, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1479459239.833358, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1479459239.833358, type 1 (EV_KEY), code 325 (BTN_TOOL_FINGER), value 0
Event: time 1479459239.833358, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 0
Event: time 1479459239.833358, -------------- SYN_REPORT ------------

chrisrobinson76, 2016-11-18:
I don't appear to have xserver-xorg-input-libinput installed.

vlasenko, 2016-11-18:
Thanks, looks like the driver is working normally.

Could you also post the output of:
`xinput list-props "Asus FTE TouchPad"`

and the output of
`synclient` 
if xinput will show that you are using Synaptics

chrisrobinson76, 2016-11-18:
Xinput:

chris@chris-X540SA:~ > xinput list-props "Asus FTE TouchPad"
Device 'Asus FTE TouchPad':
        Device Enabled (138):   1
        Coordinate Transformation Matrix (140): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        Device Accel Profile (265):     1
        Device Accel Constant Deceleration (266):       2.500000
        Device Accel Adaptive Deceleration (267):       1.000000
        Device Accel Velocity Scaling (268):    12.500000
        Synaptics Edges (269):  111, 2683, 94, 1664
        Synaptics Finger (270): 12, 15, 0
        Synaptics Tap Time (271):       180
        Synaptics Tap Move (272):       20
        Synaptics Tap Durations (273):  180, 180, 100
        Synaptics ClickPad (274):       1
        Synaptics Middle Button Timeout (275):  0
        Synaptics Two-Finger Pressure (276):    142
        Synaptics Two-Finger Width (277):       3
        Synaptics Scrolling Distance (278):     10, 10
        Synaptics Edge Scrolling (279): 1, 0, 0
        Synaptics Two-Finger Scrolling (280):   1, 0
        Synaptics Move Speed (281):     1.000000, 1.750000, 0.060588, 0.000000
        Synaptics Off (282):    2
        Synaptics Locked Drags (283):   0
        Synaptics Locked Drags Timeout (284):   5000
        Synaptics Tap Action (285):     2, 3, 0, 0, 1, 3, 0
        Synaptics Click Action (286):   1, 3, 0
        Synaptics Circular Scrolling (287):     0
        Synaptics Circular Scrolling Distance (288):    0.100000
        Synaptics Circular Scrolling Trigger (289):     0
        Synaptics Circular Pad (290):   0
        Synaptics Palm Detection (291): 0
        Synaptics Palm Dimensions (292):        5, 100
        Synaptics Coasting Speed (293): 20.000000, 50.000000
        Synaptics Pressure Motion (294):        15, 80
        Synaptics Pressure Motion Factor (295): 1.000000, 1.000000
        Synaptics Resolution Detect (296):      1
        Synaptics Grab Event Device (297):      0
        Synaptics Gestures (298):       1
        Synaptics Capabilities (299):   1, 0, 0, 1, 1, 1, 1
        Synaptics Pad Resolution (300): 1, 1
        Synaptics Area (301):   0, 0, 0, 0
        Synaptics Soft Button Areas (302):      1397, 0, 1441, 0, 0, 0, 0, 0
        Synaptics Noise Cancellation (303):     16, 16
        Device Product ID (257):        2821, 257
        Device Node (258):      "/dev/input/event9"

chrisrobinson76, 2016-11-18:
chris@chris-X540SA:~ > synclient
Parameter settings:
    LeftEdge                = 111
    RightEdge               = 2683
    TopEdge                 = 94
    BottomEdge              = 1664
    FingerLow               = 12
    FingerHigh              = 15
    MaxTapTime              = 180
    MaxTapMove              = 20
    MaxDoubleTapTime        = 180
    SingleTapTimeout        = 180
    ClickTime               = 100
    EmulateMidButtonTime    = 0
    EmulateTwoFingerMinZ    = 142
    EmulateTwoFingerMinW    = 3
    VertScrollDelta         = 10
    HorizScrollDelta        = 10
    VertEdgeScroll          = 1
    HorizEdgeScroll         = 0
    CornerCoasting          = 0
    VertTwoFingerScroll     = 1
    HorizTwoFingerScroll    = 0
    MinSpeed                = 1
    MaxSpeed                = 1.75
    AccelFactor             = 0.0605877
    TouchpadOff             = 2
    LockedDrags             = 0
    LockedDragTimeout       = 5000
    RTCornerButton          = 2
    RBCornerButton          = 3
    LTCornerButton          = 0
    LBCornerButton          = 0
    TapButton1              = 1
    TapButton2              = 3
    TapButton3              = 0
    ClickFinger1            = 1
    ClickFinger2            = 3
    ClickFinger3            = 0
    CircularScrolling       = 0
    CircScrollDelta         = 0.1
    CircScrollTrigger       = 0
    CircularPad             = 0
    PalmDetect              = 0
    PalmMinWidth            = 5
    PalmMinZ                = 100
    CoastingSpeed           = 20
    CoastingFriction        = 50
    PressureMotionMinZ      = 15
    PressureMotionMaxZ      = 80
    PressureMotionMinFactor = 1
    PressureMotionMaxFactor = 1
    ResolutionDetect        = 1
    GrabEventDevice         = 0
    TapAndDragGesture       = 1
    AreaLeftEdge            = 0
    AreaRightEdge           = 0
    AreaTopEdge             = 0
    AreaBottomEdge          = 0
    HorizHysteresis         = 16
    VertHysteresis          = 16
    ClickPad                = 1
    RightButtonAreaLeft     = 1397
    RightButtonAreaRight    = 0
    RightButtonAreaTop      = 1441
    RightButtonAreaBottom   = 0
    MiddleButtonAreaLeft    = 0
    MiddleButtonAreaRight   = 0
    MiddleButtonAreaTop     = 0
    MiddleButtonAreaBottom  = 0

ceejayrao, 2016-11-18:
On my system too, I see this issue of pointer not working. However, the reason I figured for this issue is, when mouse pointer is on webpage and the tooltip show's up, that is when tap are not recognised as click, just move the pointer to close the tooltip and then click again seem to work always.

vlasenko, 2016-11-18:
I've just pushed the commit that attempts to fix this issue. Could you update the driver and report back if this fixes the issue for you or have no effect?
https://github.com/vlasenko/hid-asus-fte-dkms#user-content-updating-to-the-latest-driver-version

vlasenko, 2016-11-18:
Oops, sorry, forgot to push. Update driver now, please

chrisrobinson76, 2016-11-18:
Sorry - it is still showing the same behaviour.

vlasenko, 2016-11-18:
Chris,

Could you describe in more detail your observations on the issue with the cursor. Does this happens on some related event or does it happens completely randomly. Does any errors are output to dmesg when this issue happens?

Thanks

chrisrobinson76, 2016-11-18:
I'll try...
The laptop boots ok and the touchpad works as a multitouch device. When the cursor stops responding, it can happen at any point randomly but I seem to be able to trigger the problem by repeatedly using a single finger on the right hand edge of the touchpad to do a vertical scroll (i.e. if I vigorously scroll the page up and down)

When the cursor stops responding the only thing that works is:
1. I can scroll the page using the right-edge of the touchpad 
2. The left and right buttons work normally 
Other multitouch features (e.g. two-finger scroll) do not work. Sometimes but not every time, if I press hard on the touchpad so it clicks (as you would to highlight text) I can then move the cursor. However it remains unresponsive to normal single finger touch to move the cursor until I switch to a console (ctrl-alt-f7) and back to X (ctrl-alt-f8). After that, it works as it should until the next unresponsive event.

It feels like its something to do with the right hand edge as this is a trigger and it continues working during the non-responsive event. Flipping in and out of the X environment must reset something to restore functions for a while.

Some possibly related messages in dmesg below. The "i2c_hid..." messages (row 1428, 3807,4827) definitely follow a non-responsive event.:
---
[    9.055967] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:633 / ret_size:0
[    9.061369] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:131 / ret_size:0
[    9.061538] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/808622C1:03/i2c-9/i2c-FTE1001:00/0018:0B05:0101.0001/input/input9
[    9.065021] hid-generic 0018:0B05:0101.0001: input,hidraw0: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
--------
[    9.183757] hid_asus_fte: module verification failed: signature and/or required key missing - tainting kernel
[    9.185585] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/808622C1:03/i2c-9/i2c-FTE1001:00/0018:0B05:0101.0001/input/input10
[    9.187505] hid-asus-fte 0018:0B05:0101.0001: input,hidraw0: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
------
[ 1102.981694] perf interrupt took too long (2515 > 2500), lowering kernel.perf_event_max_sample_rate to 50000
[ 1428.921397] i2c_hid i2c-FTE1001:00: i2c_hid_get_input: incomplete report (30/29448)
[ 2376.997117] perf interrupt took too long (5011 > 5000), lowering kernel.perf_event_max_sample_rate to 25000
[ 3807.417554] i2c_hid i2c-FTE1001:00: i2c_hid_get_input: incomplete report (30/7710)
[ 4827.276251] i2c_hid i2c-FTE1001:00: i2c_hid_get_input: incomplete report (30/12342)

Thanks!
Chris

redmcg, 2016-11-19:
@chrisrobinson76 
I wasn't able to reproduce the issue with the cursor not responding - but I could reproduce the 'incomplete report' error. I've submitted a fix for the 'incomplete report' error but I'm hoping it will also fix your cursor not responding.

Could you please perform a:
`git pull`

And run `./dev-run.sh`.

If this fixes the issue you'll want to run `./dkms-add.sh` so the fix is loaded on your next reboot.

redmcg, 2016-11-19:
Actually - I realise now that I didn't recreate the issue when I thought I had (it had happened prior to me trying to recreate it) - and I haven't fixed the 'incomplete report' error. That error happens inside hid-core (which uses the HID report from the device to setup the report size - and then discards any report from the device which is larger. This is what results in the error you are seeing).

So we'll continue to see the 'incomplete report' error (that seems to be a device problem - or there could be additional initialisation we have to do that we don't know about).

But I have made two additional changes to the driver which may resolve the cursor not responding issue:
1. Check the report id and report size before processing the report (just in case the device was sending something unexpected); and
2. Report events on all inputs in the hdev inputs list (just in case there are multiple or the one we were storing became stale)

chrisrobinson76, 2016-11-19:
Just tested it and unfortunately I'm still seeing the same issue. I checked dmesg and got the following:

[  488.845969] input: Asus FTE TouchPad as /devices/pci0000:00/808622C1:03/i2c-9/i2c-FTE1001:00/0018:0B05:0101.0002/input/input11
[  488.846390] hid-asus-fte 0018:0B05:0101.0002: input,hidraw0: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
[  685.517797] i2c_hid i2c-FTE1001:00: i2c_hid_get_input: incomplete report (30/32560)

Happy to continue testing things.
Thanks,
Chris

vlasenko, 2016-11-19:
Are you sure that the line about incomplete report happens when issue appears? I'm not sure that incomplete report is related to the issue. But if it really is, then it would be helpful for us to fix it. Unfortunately I was not able to reproduce this issue.

vlasenko, 2016-11-19:
I think we should do HID Report descriptor fixup similar to what some other HID drivers do, when it is broken. Or we could somehow disable hid subsystem to query HID Reports from the device, as this is not needed for our driver.

redmcg, 2016-11-19:
@vlasenko 
Actually I'm starting to think this could be a Synaptics issue. Initially I was thinking I might have received an unexpected report and interpreted it as a movement - but that would only affect one frame and not cause the freeze that is being described.

And looking back over the comments - https://github.com/vlasenko/hid-asus-fte-dkms/issues/9#issuecomment-261480457 shows the driver working correctly during the freeze.

@chrisrobinson76 
Can you let me know the version of Synaptics you're running - with `synclient -V`.

Also - I wonder if you can still reproduce this when Vertical Edge Scrolling is switched off:
`synclient VertEdgeScroll=0`

You can turn it back on with:
`synclient VertEdgeScroll=1`

chrisrobinson76, 2016-11-20:
Synaptics version is 1.8.99

I switched off vertical scroll as suggested, using two fingers to do vertical scroll instead. The cursor is still freezing with VertEdgeScroll=0 unfortunately.

chrisrobinson76, 2016-11-20:
@vlasenko I've looked at dmesg output after each event and they don't always follow a cursor freeze so this is possibly unrelated.

redmcg, 2016-11-20:
Does it only occur when scrolling?

redmcg, 2016-11-20:
Or is there any messages in /var/log/syslog that consistently show up when the issue occurs?

Also - could you please run:
`apt-cache policy xserver-xorg-input-synaptics`

chrisrobinson76, 2016-11-20:
No it can stop responding at any point but scrolling seems to trigger it particularly.

chris@chris-X540SA:~ > apt-cache policy xserver-xorg-input-synaptics
xserver-xorg-input-synaptics:
  Installed: (none)
  Candidate: 1.7.4-0ubuntu1
  Version table:
     1.7.4-0ubuntu1 0
        500 http://archive.ubuntu.com/ubuntu/ trusty/main amd64 Packages

redmcg, 2016-11-20:
Do you know how Synaptics was installed on your system? I think Synaptics version 1.8.99 is unstable and it doesn't appear to have been installed via your package manager (unless Mint provides it through another package).

I would suggest installing it via the package manager (`apt install xserver-xorg-input-synaptics`) but not knowing how it was installed originally - I'm not sure what impact that would have on your system.

redmcg, 2016-11-20:
OK - I just reread your first message and you're using KDE. And I think that uses the `synaptic` package (not `xserver-xorg-input-synaptics`).

Try:
`apt-cache policy synaptic`

redmcg, 2016-11-20:
Sorry - ignore that last comment. I think that `synaptic` package is for package management.

If you run:
`apt-file find synclient`

it'll tell you which package provided `synclient`. You might need to install `apt-file` first and then do an update:
```
apt install apt-file
apt-file update
```

chrisrobinson76, 2016-11-20:
@redmcg 
I didn't find anything different in /var/log/syslog than what I had already seen from dmesg.
To the best of my knowledge, the synaptics package was just whatever was set as default with the system. I personally haven't changed anything.  However, things looked inconsistent following your comments so I followed your instructions to reinstall xserver-xorg-input-synaptics and identified that there were some broken dependencies with  xorg-input-abi-20 missing and another package at the wrong version. I installed xorg-input-abi-20 and then reinstalled xserver-xorg-input-synaptics. I now have:

chris@chris-X540SA:~ > synclient -V
1.7.4

chris@chris-X540SA:~ > apt-cache policy xserver-xorg-input-synaptics
xserver-xorg-input-synaptics:
  Installed: 1.7.4-0ubuntu1
  Candidate: 1.7.4-0ubuntu1
  Version table:
 *** 1.7.4-0ubuntu1 0
        500 http://archive.ubuntu.com/ubuntu/ trusty/main amd64 Packages
        100 /var/lib/dpkg/status

Since doing that (about 15 minutes ago) the pointer has not frozen. I'll keep using the system and report back tomorrow if this is now working consistently. My feeling is that it is now. Sorry in advance if this has all been due to some quirk in my system but thanks both for helping me find a problem.

Chris

redmcg, 2016-11-20:
Thanks Chris. Hopefully it continues to work OK.

And no apology needed. This resulted in me taking a closer look at my code and I think the driver is better for it.

I'll close the issue but please do reopen it if the problem occurs again.

commit e7b8e482 (2016-11-18): Attempt to fix #9

Try to sync frame, hopefully this can help with out of sync between driver
and input processing libs.
closed 2016-11-20 by redmcg