larixer/hid-asus-dkms #12: Failed to reset device

OBrezhniev, 2016-11-20:
Hi,
I'm running Fedora 24 on Asus Zenbook Pro UX501VW.
There are some errors on driver load.
```
uname -a
Linux notebook.local 4.8.8-200.fc24.x86_64 #1 SMP Tue Nov 15 19:41:51 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```
```
dmesg | grep hid
[    1.022974] hidraw: raw HID events driver (C) Jiri Kosina
[    1.022999] usbcore: registered new interface driver usbhid
[    1.023000] usbhid: USB HID core driver
[    2.284137] hid-generic 0003:0458:0185.0001: hiddev0,hidraw0: USB HID v1.10 Device [MOSART Semi. Wireless Mouse] on usb-0000:00:14.0-6/input0
[    2.338979] hid-generic 0003:0458:0185.0002: input,hiddev0,hidraw1: USB HID v1.10 Mouse [MOSART Semi. Wireless Mouse] on usb-0000:00:14.0-6/input1
[   10.229025] i2c_hid i2c-FTE1001:00: failed to reset device.
[   11.274175] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:633 / ret_size:0
[   11.277580] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:131 / ret_size:0
[   11.278586] hid-generic 0018:0B05:0101.0003: input,hidraw2: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
[   11.286109] hid_asus_fte: loading out-of-tree module taints kernel.
[   11.286169] hid_asus_fte: module verification failed: signature and/or required key missing - tainting kernel
```
```
lsmod | grep hid
i2c_hid                20480  0
hid_asus_fte           16384  0
```
```
modprobe -r hid_asus_fte hid_generic
modprobe: FATAL: Module hid_generic is builtin.
```
```
xinput list
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ MOSART Semi. Wireless Mouse               id=11   [slave  pointer  (2)]
⎜   ↳ FTE1001:00 0B05:0101                      id=12   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Video Bus                                 id=7    [slave  keyboard (3)]
    ↳ Video Bus                                 id=8    [slave  keyboard (3)]
    ↳ Sleep Button                              id=9    [slave  keyboard (3)]
    ↳ USB2.0 HD UVC WebCam                      id=10   [slave  keyboard (3)]
    ↳ Asus WMI hotkeys                          id=13   [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard              id=14   [slave  keyboard (3)]
    ↳ MOSART Semi. Wireless Mouse               id=15   [slave  keyboard (3)]
```

Let me know if you need more debug info.

redmcg, 2016-11-21:
This line:
`modprobe: FATAL: Module hid_generic is builtin`
means that the hid_generic module has been built in to your kernel. So the `modprobe -r` command won't work. Instead you'll need to manually unbind the device from the `hid_generic` driver and then bind it to the new `hid-asus-fte` driver.

You should be able to do that with the following commands:
```
sudo bash -c 'echo 0018:0B05:0101.0003 > /sys/bus/hid/drivers/hid-generic/unbind'
sudo bash -c 'echo 0018:0B05:0101.0003 > /sys/bus/hid/drivers/hid-asus-fte/bind'
```

OBrezhniev, 2016-11-21:
Thanks @redmcg, that works!
I only had to change id string to 0018:0B05:0101.0007.
How can I make it persistent?

redmcg, 2016-11-21:
I'm not 100% sure (as I haven't tested with `hid_generic` compiled in to the kernel) but if you add the module to your `initrd` - I'm hoping the kernel will pick it first.

I haven't used Fedora but I think you can use `dracut` to add a module. During my research I found this:
https://www.kernel.org/pub/linux/utils/boot/dracut/dracut.html#_adding_kernel_modules

So I think the command would be:
`dracut -fv --add-drivers hid-asus-fte`

but you might know better than I do!

redmcg, 2016-11-21:
@OBrezhniev How's it going? Are you happy for us to close this issue?

We've created a new issue (#16) to address the errors you saw in your log so I just want to make sure this is working for you with `hid_generic` compiled in to the kernel.

OBrezhniev, 2016-11-21:
I tried adding driver to initrd with dracut, but it doesn't make kernel choose hid-asus-fte.

OBrezhniev, 2016-11-21:
I can create init script to rebind device, but may be you know a a better way.

redmcg, 2016-11-21:
I've just modified the `dkms.conf` in my fork of this repo. You can find it here: https://github.com/redmcg/hid-asus-fte-dkms

Instead of unloading the `hid_generic` module (which won't work in your setup) it tries to unbind the device from its current driver before loading the new module.

If you like you can try cloning from my repo and running `dkms-add.sh`. Let me know if this resolves the issue.

OBrezhniev, 2016-11-22:
No, it doesn't work for me again. Tomorrow I'll try to write udev rule using your script from `dkms.conf`.

vlasenko, 2016-11-22:
@OBrezhniev Could you try to update the driver from source:
https://github.com/vlasenko/hid-asus-dkms#user-content-updating-to-the-latest-driver-version

There were some changes added to drop modprobe hacks and use module.alias instead for the device

OBrezhniev, 2016-11-22:
Ok, I'll try in a few hours.

OBrezhniev, 2016-11-22:
@vlasenko I've updated, but ./dkms-add.sh failed as I already have `hid-asus` module bundled with fedora kernel.

OBrezhniev, 2016-11-22:
Will try revision before refactoring.

vlasenko, 2016-11-22:
@OBrezhniev Actually it is valid for DKMS to replace modules with the same name. Why do you think it fails? Is there any indication in dmesg that hid-generic is used for the device instead of hid-asus?

OBrezhniev, 2016-11-22:
With build of e3f9616 I get:
```
$ dmesg | grep -i hid
[    1.023195] hidraw: raw HID events driver (C) Jiri Kosina
[    1.023221] usbcore: registered new interface driver usbhid
[    1.023221] usbhid: USB HID core driver
[    2.292463] hid-generic 0003:0458:0185.0001: hiddev0,hidraw0: USB HID v1.10 Device [MOSART Semi. Wireless Mouse] on usb-0000:00:14.0-6/input0
[    2.348221] hid-generic 0003:0458:0185.0002: input,hiddev0,hidraw1: USB HID v1.10 Mouse [MOSART Semi. Wireless Mouse] on usb-0000:00:14.0-6/input1
[    4.246217] hid_asus_fte: loading out-of-tree module taints kernel.
[    4.257678] hid_asus_fte: module verification failed: signature and/or required key missing - tainting kernel
[    9.716301] i2c_hid i2c-FTE1001:00: failed to reset device.
[   10.761558] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:633 / ret_size:0
[   10.764991] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:131 / ret_size:0
[   10.765959] hid-generic 0018:0B05:0101.0003: input,hidraw2: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
```

OBrezhniev, 2016-11-22:
Same thing on master:
```
$ dmesg | grep hid
[    1.021255] hidraw: raw HID events driver (C) Jiri Kosina
[    1.021281] usbcore: registered new interface driver usbhid
[    1.021281] usbhid: USB HID core driver
[    2.284196] hid-generic 0003:0458:0185.0001: hiddev0,hidraw0: USB HID v1.10 Device [MOSART Semi. Wireless Mouse] on usb-0000:00:14.0-6/input0
[    2.339602] hid-generic 0003:0458:0185.0002: input,hiddev0,hidraw1: USB HID v1.10 Mouse [MOSART Semi. Wireless Mouse] on usb-0000:00:14.0-6/input1
[    4.252223] hid_asus: loading out-of-tree module taints kernel.
[    4.284591] hid_asus: module verification failed: signature and/or required key missing - tainting kernel
[    9.715658] i2c_hid i2c-FTE1001:00: failed to reset device.
[   10.760849] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:633 / ret_size:0
[   10.764250] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:131 / ret_size:0
[   10.765174] hid-generic 0018:0B05:0101.0003: input,hidraw2: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
```
Previous time I haven't tried to reboot because of an error message from `./dkms-add.sh` like this:
```
...
DKMS: install completed.
Rebinding 0018:0B05:0101.0005 to hid-asus
sh: line 0: echo: write error: No such device
sh: line 0: echo: write error: No such device
```
I had checked /sys/bus/hid/drivers/hid-asus/bind - and it hadn't existed, because of bundled hid-asus was loaded.

OBrezhniev, 2016-11-22:
In both cases running `./dev-attach.sh` after reboot rebinds device successfully.

vlasenko, 2016-11-23:
Yes, I'm not sure we can solve this in DKMS driver. We are not aware of reliable way to specify on boot that "hid-asus" should be used for device and not "hid-generic". Current way of trying to catch device early using ACPI ID: "FTE1*" does not guarantee that the device will be attached to our driver. When this code will be submitted into Linux kernel we will modify hid core as well and flag that the device has special driver, so that hid core don't load hid-generic for the device. 

So for your case the only work around we can propose for DKMS driver is manually rebinding the driver for now after boot.

If you happen to find another solution, please let us know.

Thanks

redmcg, 2016-11-25:
@OBrezhniev 

Can you please try the following udev rule:
```
ACTION=="add" ENV{MODALIAS}=="hid:b0018g0001v00000B05p00000101", RUN+="/bin/sh -c 'modprobe -i hid-asus; D=`basename %p`; echo $D > /sys/bus/hid/drivers/hid-generic/unbind; echo $D > /sys/bus/hid/drivers/hid-asus/bind'"
```

I tested this in a file called:
`/etc/udev/rules.d/41-hid-asus.rules`

redmcg, 2016-11-26:
I tested the above rule with a number of different kernel configurations and it worked for all except when the pre-existing `hid-asus.c` (the current asus driver that supports just keyboard) was compiled in to the kernel. This was because `modprobe` didn't work - but `insmod` did. So I changed the rule to be:
```
ACTION=="add" ENV{MODALIAS}=="hid:b0018g0001v00000B05p00000101", RUN+="/bin/sh -c 'kname=`uname -r`; insmod `find /lib/modules/$kname -name hid-asus.ko ! -wholename /lib/modules/$kname/kernel/drivers/hid/hid-asus.ko`; D=`basename %p`; echo $D > /sys/bus/hid/drivers/hid-generic/unbind; echo $D > /sys/bus/hid/drivers/hid-asus/bind'"
```

This worked for all kernel configs I tested.

@vlasenko I've made a commit to my fork which creates a udev file with the above content when you run `dkms-add.sh` (and removes it on `dkms-remove.sh`). Let me know if you want me to push it here.

vlasenko, 2016-11-26:
@redmcg Great! Brendan, push it here as well, please.

redmcg, 2016-11-27:
@vlasenko Done. Note that I removed the `MODULE_ALIAS` declaration - just because it shouldn't be needed with the udev rule and it makes the DKMS pretty much the same as what we plan submit to the kernel. I'm happy for it to be added back if you find its needed though.

vlasenko, 2016-11-27:
@redmcg Yes, sure, I think MODULE_ALIAS is not needed, when udev rule is used. Thanks!

OBrezhniev, 2016-11-27:
Thanks @redmcg, @vlasenko for your great work! Now everything works super fine!

redmcg, 2016-11-27:
That's great to hear! Thanks for helping us test.

closed 2016-11-27 by redmcg