larixer/hid-asus-dkms #14: Reorder asus_setup_input()

bentiss, 2016-11-21:
input_mt_init_slots(input, MAX_CONTACTS, 0); should be called last, with proper flags.

This way, you don't need to set most of the bits in asus_setup_input() (BTN_LEFT, BTN_TOOL_*, etc) and you don't need to set the non MT axes.

The correct order is to set the ABS_MT bits, to the correct ranges, and call input_mt_init_slots(input, MAX_CONTACTS, INPUT_MT_POINTER);

While talking about the mutlitouch protocol, in asus_report_input(), instead of calling input_mt_report_pointer_emulation(input, true);, it's better to call input_mt_sync_frame(input), so that if you need in the future handle more flags (kernel tracking or dropping of slots), you won't have to change this part.

closed 2016-11-21 by vlasenko

redmcg, 2016-11-21:
Thanks for that. I used the `Documentation/input/multi-touch-protocol.txt` for understanding how to use the multi-touch API. And where there were gaps I used `hid-magicmouse.c` as a reference.

Unfortunately the documentation didn't mention either `input_mt_init_slots` or `input_mt_sync_frame` so I followed `hid-magicmouse.c` (which used zero as its flag on `input_mt_init_slots` and it didn't use `input_mt_sync_frame`).

I guess my best reference at the moment (apart from yourself of course) is the source code in `drivers/input/input-mt.c`. But if you can recommend any other (preferably online) references I'd really appreciate it.

bentiss, 2016-11-21:
> I guess my best reference at the moment (apart from yourself of course) is the source code in drivers/input/input-mt.c. But if you can recommend any other (preferably online) references I'd really appreciate it.

Unfortunately, there is no other documentation. The documentation of the multitouch protocol should probably get updated (patches welcome :-P), but the reference is the actual source.