larixer/hid-asus-dkms #17: Right click not working

ilgare, 2016-11-21:
Hi there,
This is on kernel 4.8.10 (also tested on 4.8.8). The device (on an Asus TP200SA) is reported as:

FTE1000:00 0B05:0101 

When I install the driver (running dkms-add) and reboot, it is loaded but the touchpad is still recognized as a mouse. Running the rmmod/insmod sequence as in dev-run.sh, I can bring it up as a touchpad. I get the two-finger scroll bu this time right-click stops working. xev reports the right click as Button 1.

redmcg, 2016-11-21:
@ilgare Thanks for the report.

Could you please run:
```
xinput
xinput list-props "Asus FTE TouchPad"
```

And please provide the output.

ilgare, 2016-11-22:
Here they are:

```
bash-4.3$ xinput
 Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Logitech M505/B605                      	id=11	[slave  pointer  (2)]
⎜   ↳ Logitech K340                           	id=12	[slave  pointer  (2)]
⎜   ↳ Asus FTE TouchPad                       	id=13	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Asus Wireless Radio Control             	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ USB2.0 VGA UVC WebCam                   	id=10	[slave  keyboard (3)]
    ↳ Asus WMI hotkeys                        	id=15	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=16	[slave  keyboard (3)]
∼ SIS0457:00 0457:11CF                    	id=14	[floating slave]
bash-4.3$ xinput list-props "Asus FTE TouchPad"
Device 'Asus FTE TouchPad':
	Device Enabled (138):	1
	Coordinate Transformation Matrix (140):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (270):	1
	Device Accel Constant Deceleration (271):	2.500000
	Device Accel Adaptive Deceleration (272):	1.000000
	Device Accel Velocity Scaling (273):	12.500000
	Synaptics Edges (557):	111, 2683, 94, 1664
	Synaptics Finger (558):	12, 15, 0
	Synaptics Tap Time (559):	180
	Synaptics Tap Move (560):	145
	Synaptics Tap Durations (561):	180, 100, 100
	Synaptics ClickPad (562):	1
	Synaptics Middle Button Timeout (563):	0
	Synaptics Two-Finger Pressure (564):	142
	Synaptics Two-Finger Width (565):	3
	Synaptics Scrolling Distance (566):	66, 66
	Synaptics Edge Scrolling (567):	0, 0, 0
	Synaptics Two-Finger Scrolling (568):	1, 0
	Synaptics Move Speed (569):	1.000000, 1.750000, 0.060588, 0.000000
	Synaptics Off (570):	0
	Synaptics Locked Drags (571):	0
	Synaptics Locked Drags Timeout (572):	5000
	Synaptics Tap Action (573):	0, 0, 0, 0, 1, 2, 3
	Synaptics Click Action (574):	1, 3, 2
	Synaptics Circular Scrolling (575):	0
	Synaptics Circular Scrolling Distance (576):	0.100000
	Synaptics Circular Scrolling Trigger (577):	0
	Synaptics Circular Pad (578):	0
	Synaptics Palm Detection (579):	0
	Synaptics Palm Dimensions (580):	5, 100
	Synaptics Coasting Speed (581):	20.000000, 50.000000
	Synaptics Pressure Motion (582):	15, 80
	Synaptics Pressure Motion Factor (583):	1.000000, 1.000000
	Synaptics Grab Event Device (584):	0
	Synaptics Gestures (585):	1
	Synaptics Capabilities (586):	1, 0, 0, 1, 1, 1, 1
	Synaptics Pad Resolution (587):	1, 1
	Synaptics Area (588):	0, 0, 0, 0
	Synaptics Soft Button Areas (589):	0, 0, 0, 0, 0, 0, 0, 0
	Synaptics Noise Cancellation (590):	16, 16
	Device Product ID (259):	2821, 257
	Device Node (260):	"/dev/input/event7"
```

redmcg, 2016-11-22:
Could you please try running:
```
xinput set-prop "Asus FTE TouchPad" "Synaptics Soft Button Areas" 1397 0 1441 0 0 0 0 0
```

and let me know if that fixes the issue?

ilgare, 2016-11-22:
Bingo! Scroll+right click are working now.

By the way I didn't have the opportunity to thank you guys at the kernel bugzilla thread, so let me do it here. I appreciate the great work you've done, thanks a lot.

redmcg, 2016-11-22:
No problem. Thanks for helping us test.

closed 2016-11-22 by redmcg