larixer/hid-asus-dkms #18: To restart the touchpad does not work

ligomat, 2016-11-21:
I apologize for my English but I'm Italian. I have a asus x556u, I followed the guide and the touchpad but it works to restart does not work and whenever I have to install the drivers from the terminal. I also had to download the file as .zip and then do it by the book because it does not make me add repository saying that is not valid. How can I fix?

redmcg, 2016-11-22:
@ligomat Thanks for the report.

Could you please copy and paste the error you are seeing?

ligomat, 2016-11-22:
@redmcg  the error:
`Cloning into 'hid-asus-fte-dkms'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.`

redmcg, 2016-11-22:
Try using https instead:
`git clone https://github.com/vlasenko/hid-asus-fte-dkms.git`

ligomat, 2016-11-22:
@redmcg  perfect, it works!! Thanks !!! :)

redmcg, 2016-11-22:
Is the touchpad working after restart?

vlasenko, 2016-11-22:
@ligomat In case of touchpad is not working after reboot, please try to update the driver from source, there were some relevant changes today:
https://github.com/vlasenko/hid-asus-dkms#user-content-updating-to-the-latest-driver-version

vlasenko, 2016-11-23:
Closing this one because of no issue author activity. Please reopen this issue if you still have any problems with the driver.

closed 2016-11-23 by vlasenko