larixer/hid-asus-dkms #19: Doesn't work on Fedora 25 with Wayland

thomasnaudin, 2016-11-22:
I've got an Asus UX501VW.

```
# dmesg | grep -i mouse
[    1.065740] mousedev: PS/2 mouse device common for all mice
[    5.663243] hid-generic 0018:0B05:0101.0003: input,hidraw2: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
```

The module compiles and loads :

```
# lsmod | grep asus
hid_asus_fte           16384  0
asus_nb_wmi            24576  0
asus_wmi               28672  1 asus_nb_wmi
sparse_keymap          16384  1 asus_wmi
rfkill                 24576  8 asus_wmi,bluetooth,cfg80211
asus_wireless          16384  0
wmi                    16384  3 asus_wmi,mxm_wmi,nouveau
video                  40960  4 asus_wmi,int3406_thermal,nouveau,i915
```

vlasenko, 2016-11-22:
@thomasnaudin Thanks for reporting an issue.

Seems you are not using the latest source code. It has been changed today.

Please try to update the driver from source:
https://github.com/vlasenko/hid-asus-dkms#user-content-updating-to-the-latest-driver-version

There were some changes added to drop modprobe hacks and use module.alias instead for the device. Perhaps they will be helpful for your case.

thomasnaudin, 2016-11-22:
Works perfectly now, sorry, I didn't check today's version.
Thanks a lot, I've been hoping for a working touchpad for some months now :)

vlasenko, 2016-11-22:
Cool! Thank you very much for testing and reporting back!

closed 2016-11-22 by vlasenko