larixer/hid-asus-dkms #20: Issues on Ubuntu

Quinton92, 2016-11-23:
So, I'm fairly new to Linux in general, and I am sorry to bother you. I recently have been trying different distros like Linux Mint KDE, Linux Mint Cinnamon, elementary os, and zorin os. On all of these, your patch has worked flawlessly. I just recently installed Ubuntu 16.04, and went to apply your patch. First I made sure that dkms was installed via terminal, then I cloned your git folder, and ran the patch like usual with ./dkms-add.sh. I am receiving some errors. 

This is my error.

./dkms-add.sh
Error! There are no instances of module: asus-fte
1.0 located in the DKMS tree.
Error! There are no instances of module: asus
1.0 located in the DKMS tree.

Creating symlink /var/lib/dkms/asus/1.0/source ->
                 /usr/src/asus-1.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make KERNELRELEASE=4.4.0-47-generic -C /lib/modules/4.4.0-47-generic/build M=/var/lib/dkms/asus/1.0/build/src hid-asus.ko....
cleaning build area....

DKMS: build completed.

hid-asus.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.4.0-47-generic/updates/dkms/

depmod....

Backing up initrd.img-4.4.0-47-generic to /boot/initrd.img-4.4.0-47-generic.old-dkms
Making new initrd.img-4.4.0-47-generic
(If next boot fails, revert to initrd.img-4.4.0-47-generic.old-dkms image)
update-initramfs.......

DKMS: install completed.
modprobe: ERROR: could not insert 'hid_asus': Required key not available
Rebinding 0018:0B05:0101.0002 to hid-asus
sh: echo: I/O error
sh: 1: cannot create /sys/bus/hid/drivers/hid-asus/bind: Directory nonexistent

Do you have any advice for me?

redmcg, 2016-11-23:
You can ignore this bit:
```
Error! There are no instances of module: asus-fte
1.0 located in the DKMS tree.
Error! There are no instances of module: asus
1.0 located in the DKMS tree.
```

This is code looking for the older names that this module used. We should probably change our script to suppress those errors.

I think the issue is this bit:
`modprobe: ERROR: could not insert 'hid_asus': Required key not available`

It looks like you are using secure boot. You won't be able to run third-party (aka 'out of tree') kernel modules. This is because for secure boot to work all the modules have to be signed - and given our module is third party is hasn't been signed.

If you want to use this module you'll need to disable secure boot.

Quinton92, 2016-11-23:
Awesome thank you! I will disable secure boot. I thought i did that in the ubuntu installation. Sorry for the bother!

redmcg, 2016-11-23:
No problem. Thanks for helping us test. Let me know if this does resolve the problem and then I will close this issue.

Quinton92, 2016-11-23:
Yes it did! I had that error before on a different distro, and fixed it by disabling secure boot, but i was 99% sure that I had disabled secure boot before receiving the error message again. Thank you for your time.

redmcg, 2016-11-23:
Excellent - thanks for confirmation.

closed 2016-11-23 by redmcg