larixer/hid-asus-dkms #22: Fail during install

itped, 2016-11-24:
Hello, I am getting an error message that I do not know how to deal with. It seems that some driver is not being recognized and because of that the dev-attach.sh script fails. I have an UX501VW. I appreciate your help! Please let me know if you need some more infos. 

```
Creating symlink /var/lib/dkms/asus/1.0/source ->
                 /usr/src/asus-1.0
teste
DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=4.8.10-1-ARCH -C /usr/lib/modules/4.8.10-1-ARCH/build M=/var/lib/dkms/asus/1.0/build/src hid-asus.ko...
cleaning build area...
Kernel cleanup unnecessary for this kernel.  Skipping...

DKMS: build completed.

hid-asus.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/4.8.10-1-ARCH/updates/

depmod...

DKMS: install completed.
Rebinding to hid-asus
sh: /sys/bus/hid/drivers/hid-generic/unbind: No such file or directory
sh: line 0: echo: write error: No such device <>

```

```
dmesg | grep hid
[    1.877578] hidraw: raw HID events driver (C) Jiri Kosina
[    2.378520] usbcore: registered new interface driver usbhid
[    2.378522] usbhid: USB HID core driver
[    2.406470] hid-multitouch 0003:0457:11D1.0001: input,hiddev0,hidraw0: USB HID v1.11 Device [USBest Technology SiS HID Touch Controller] on usb-0000:00:14.0-12/input0
[   55.355727] hid_asus: loading out-of-tree module taints kernel.
```

```
lsmod | grep hid
hid_asus               16384  0
hid_multitouch         20480  0
usbhid                 49152  0
mac_hid                16384  0
i2c_hid                20480  0
hid                   114688  4 i2c_hid,hid_asus,usbhid,hid_multitouch
usbcore               208896  5 uvcvideo,usbhid,xhci_pci,btusb,xhci_hcd
```

```
xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ USBest Technology SiS HID Touch Controller	id=10	[slave  pointer  (2)]
⎜   ↳ Elan Touchpad                           	id=12	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ USB2.0 HD UVC WebCam                    	id=11	[slave  keyboard (3)]
    ↳ Asus WMI hotkeys                        	id=13	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=14	[slave  keyboard (3)]
```

redmcg, 2016-11-25:
Under `xinput list` I can see your Touchpad is listed as `Elan Touchpad`. So I suspect your laptop is using a different Touchpad.

To confirm - please run:
```
basename `ls -d /sys/bus/i2c/drivers/i2c_hid/i*`
```

You should get the output of:
`i2c-FTE1001:00`

itped, 2016-11-25:
There isn't any file starting with the i in /sys/bus/i2c/drivers/i2c_hid folder:

```
basename `ls -d /sys/bus/i2c/drivers/i2c_hid/i*`
ls: cannot access '/sys/bus/i2c/drivers/i2c_hid/i*': No such file or directory
basename: missing operand
Try 'basename --help' for more information
```

I am using  xf86-input-libinput driver.

redmcg, 2016-11-25:
I believe you have an ElanTech Touchpad. This is not supported by our driver. You can run:
```
basename `ls -d /sys/bus/acpi/devices/{FT*,EL*} 2> /dev/null`
```

to confirm.

The ElanTech devices usually begin with `ELAN`

itped, 2016-11-25:
Yeah, you are right. I've got an ELAN driver. Thanks a lot for your help!

closed 2016-11-25 by itped