larixer/hid-asus-dkms #26: Error when trying the driver with Asus K501UW

serguitus, 2016-11-28:
Hi Victor,

First of all, thanks a lot for your hard work. I'm trying your driver but I'm getting the following error when executing the ```./dev-run.sh``` command
```make: Entering directory '/usr/src/linux-headers-4.4.0-45-generic'
arch/x86/Makefile:148: CONFIG_X86_X32 enabled but no binutils support
Makefile:676: Cannot use CONFIG_CC_STACKPROTECTOR_STRONG: -fstack-protector-strong not supported by compiler
make: *** No rule to make target 'driver/hid-asus-fte-dkms-master/src'.  Stop.
make: Leaving directory '/usr/src/linux-headers-4.4.0-45-generic'
make: Entering directory '/usr/src/linux-headers-4.4.0-45-generic'
arch/x86/Makefile:148: CONFIG_X86_X32 enabled but no binutils support
Makefile:676: Cannot use CONFIG_CC_STACKPROTECTOR_STRONG: -fstack-protector-strong not supported by compiler
make: *** No rule to make target 'driver/hid-asus-fte-dkms-master/src'.  Stop.
make: Leaving directory '/usr/src/linux-headers-4.4.0-45-generic'
[sudo] password for sergio: 
rmmod: ERROR: Module hid_asus_fte is not currently loaded
insmod: ERROR: could not load module src/hid-asus-fte.ko: No such file or directory
sergio@sergios:~/Touchpad driver/hid-asus-fte-dkms-master$ 
```
I'm running Kubuntu 16.04 with kernel 4.4.0-45 generic x86_64
xinput output is
```
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ FTE1001:00 0B05:0101                      id=11   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]

```
What should I do with that compiler error? is it possible to fix this or I must install a different kernel?

redmcg, 2016-11-28:
Oddly (according to this http://askubuntu.com/questions/367838/compiling-error-while-installing-realtek-rtl8111e-in-64-bit-13-10-config-x86-x) you can't compile a driver when there is a space in the path name.

I tried this myself and got the same error that you are seeing. So just try renaming 'Touchpad driver' to 'Touchpad_driver'.

vlasenko, 2016-11-28:
This is odd. @redmcg perhaps we should add notice in README.md, that path to the driver should not contain spaces?

serguitus, 2016-11-28:
Wow! that's odd. I will make sure to never name folders with spaces again :)
Thanks for your fast response. You can add the K501U to the fixed list. This works like a charm :)
I guess we can close this as fixed.

closed 2016-11-28 by serguitus