larixer/hid-asus-dkms #27: Touchpad not recognized as "Touch" by Unity (Ubuntu 16.04)

alm76, 2016-11-30:
Hi,
i've an Asus K501U, with the FTE101 touchpad, thank you for your driver, that now works well, but it's not recognized as "touchpad" under "System settings" -> "Mouse&Touchpad", so I can't configure the touchpad behavior.
[image]

my xinput:
$ xinput 
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Dell Premium USB Optical Mouse          	id=10	[slave  pointer  (2)]
⎜   ↳ Asus TouchPad                           	id=12	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ USB2.0 VGA UVC WebCam                   	id=11	[slave  keyboard (3)]
    ↳ Asus WMI hotkeys                        	id=13	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=14	[slave  keyboard (3)]

and my xinput --list-props 12
$ xinput --list-props 12
Device 'Asus TouchPad':
	Device Enabled (137):	1
	Coordinate Transformation Matrix (139):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	libinput Tapping Enabled (290):	0
	libinput Tapping Enabled Default (291):	0
	libinput Tapping Drag Enabled (292):	1
	libinput Tapping Drag Enabled Default (293):	1
	libinput Tapping Drag Lock Enabled (294):	0
	libinput Tapping Drag Lock Enabled Default (295):	0
	libinput Accel Speed (273):	0.000000
	libinput Accel Speed Default (274):	0.000000
	libinput Natural Scrolling Enabled (278):	0
	libinput Natural Scrolling Enabled Default (279):	0
	libinput Send Events Modes Available (257):	1, 1
	libinput Send Events Mode Enabled (258):	0, 0
	libinput Send Events Mode Enabled Default (259):	0, 0
	libinput Left Handed Enabled (280):	0
	libinput Left Handed Enabled Default (281):	0
	libinput Scroll Methods Available (282):	1, 1, 0
	libinput Scroll Method Enabled (283):	1, 0, 0
	libinput Scroll Method Enabled Default (284):	1, 0, 0
	libinput Click Methods Available (296):	1, 1
	libinput Click Method Enabled (297):	1, 0
	libinput Click Method Enabled Default (298):	1, 0
	libinput Disable While Typing Enabled (299):	1
	libinput Disable While Typing Enabled Default (300):	1
	Device Node (260):	"/dev/input/event8"
	Device Product ID (261):	2821, 257
	libinput Drag Lock Buttons (289):	<no items>
	libinput Horizonal Scroll Enabled (262):	1

Tranks,
Alessio

alm76, 2016-11-30:
I solved the issue by myself, so I'm writing just for other guy with the same issue.
The solution:
```
sudo apt remove xserver-xorg-input-libinput
sudo apt install xserver-xorg-input-synaptics
```

and
Unity recognize only Synaptics touchpads, no libinput, so you have to remove the **xserver-xorg-input-libinput** package and must be sure to have installed **xserver-xorg-input-synaptics**

Now it's working perfectly!! This is my "Mouse&Touchpad" now :smiley:  :
[image]

Thank you for your precious work!

closed 2016-11-30 by alm76

vlasenko, 2016-11-30:
Cool, thanks for reporting this!