larixer/hid-asus-dkms #28: Vertical/Horizontal scroll mixing

yigals, 2016-11-30:
Horizontal scrolling occurs on relevant items even when vertical scroll is in progress, and vice-versa.

redmcg, 2016-12-01:
This sounds like a problem with your Xorg input driver. Can you please run the following two commands and provide their output:
```
xinput
xinput list-props "Asus TouchPad"
```

yigals, 2016-12-01:
~$ xinput
⎡ Virtual core pointer                     id=2 [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer               id=4 [slave  pointer  (2)]
⎜   ↳ MOSART Semi. 2.4G Keyboard Mouse         id=16 [slave  pointer  (2)]
⎜   ↳ Asus FTE TouchPad                       id=12 [slave  pointer  (2)]
⎣ Virtual core keyboard                   id=3 [master keyboard (2)]
    ↳ Virtual core XTEST keyboard             id=5 [slave  keyboard (3)]
    ↳ Power Button                             id=6 [slave  keyboard (3)]
    ↳ Asus Wireless Radio Control             id=7 [slave  keyboard (3)]
    ↳ Video Bus                               id=8 [slave  keyboard (3)]
    ↳ Video Bus                               id=9 [slave  keyboard (3)]
    ↳ Sleep Button                             id=10 [slave  keyboard (3)]
    ↳ USB Camera                               id=13 [slave  keyboard (3)]
    ↳ Asus WMI hotkeys                         id=14 [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard             id=15 [slave  keyboard (3)]
    ↳ MOSART Semi. 2.4G Keyboard Mouse         id=11 [slave  keyboard (3)]

~$ xinput list-props "Asus TouchPad"
unable to find device Asus TouchPad

redmcg, 2016-12-01:
Looks like you're running an older version of the driver.

Please perform a `git pull` and try `xinput list-props "Asus TouchPad"` again

redmcg, 2016-12-04:
I'll close this - but please open again if the new version of the driver does not resolve your issue.

yigals, 2016-12-04:
sorry about the delays... getting married in 3 days :) I'll check again
afterwards.

vlasenko, 2016-12-05:
Now this is valid reason for delay ;)

yigals, 2016-12-14:
and... I'm back.
I updated the driver but the problem persists. should I open another issue?
anyway,

~$ xinput
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ MOSART Semi. 2.4G Keyboard Mouse        	id=12	[slave  pointer  (2)]
⎜   ↳ Asus TouchPad                           	id=14	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Asus Wireless Radio Control             	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Video Bus                               	id=9	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=10	[slave  keyboard (3)]
    ↳ MOSART Semi. 2.4G Keyboard Mouse        	id=11	[slave  keyboard (3)]
    ↳ USB Camera                              	id=13	[slave  keyboard (3)]
    ↳ Asus WMI hotkeys                        	id=15	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=16	[slave  keyboard (3)]
~$ xinput list-props "Asus TouchPad"
Device 'Asus TouchPad':
	Device Enabled (133):	1
	Coordinate Transformation Matrix (135):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (266):	1
	Device Accel Constant Deceleration (267):	2.500000
	Device Accel Adaptive Deceleration (268):	1.000000
	Device Accel Velocity Scaling (269):	12.500000
	Synaptics Edges (292):	111, 2683, 94, 1664
	Synaptics Finger (293):	12, 15, 0
	Synaptics Tap Time (294):	180
	Synaptics Tap Move (295):	145
	Synaptics Tap Durations (296):	180, 100, 100
	Synaptics ClickPad (297):	1
	Synaptics Middle Button Timeout (298):	0
	Synaptics Two-Finger Pressure (299):	142
	Synaptics Two-Finger Width (300):	3
	Synaptics Scrolling Distance (301):	-66, -66
	Synaptics Edge Scrolling (302):	0, 0, 0
	Synaptics Two-Finger Scrolling (303):	1, 1
	Synaptics Move Speed (304):	1.000000, 1.750000, 0.060588, 0.000000
	Synaptics Off (305):	2
	Synaptics Locked Drags (306):	0
	Synaptics Locked Drags Timeout (307):	5000
	Synaptics Tap Action (308):	2, 3, 0, 0, 1, 3, 0
	Synaptics Click Action (309):	1, 3, 0
	Synaptics Circular Scrolling (310):	0
	Synaptics Circular Scrolling Distance (311):	0.100000
	Synaptics Circular Scrolling Trigger (312):	0
	Synaptics Circular Pad (313):	0
	Synaptics Palm Detection (314):	0
	Synaptics Palm Dimensions (315):	5, 100
	Synaptics Coasting Speed (316):	20.000000, 50.000000
	Synaptics Pressure Motion (317):	15, 80
	Synaptics Pressure Motion Factor (318):	1.000000, 1.000000
	Synaptics Resolution Detect (319):	1
	Synaptics Grab Event Device (320):	0
	Synaptics Gestures (321):	1
	Synaptics Capabilities (322):	1, 0, 0, 1, 1, 1, 1
	Synaptics Pad Resolution (323):	1, 1
	Synaptics Area (324):	0, 0, 0, 0
	Synaptics Soft Button Areas (325):	1397, 0, 1441, 0, 0, 0, 0, 0
	Synaptics Noise Cancellation (326):	16, 16
	Device Product ID (256):	2821, 257
	Device Node (257):	"/dev/input/event11"

redmcg, 2016-12-14:
I'll reopen. But I can't see any problem with those settings. Can you please post the output of:
`synclient`

But I should also make sure I understand what you mean by this:
> Horizontal scrolling occurs on relevant items even when vertical scroll is in progress, and vice-versa

Do you mean:
1. vertical scroll happens when you expect horizontal scrolling; or
2. vertical scroll can happen at the same time as horizontal scrolling

Because number 2 appears to be a feature of the Synaptics Xorg Input Driver.

yigals, 2016-12-16:
It is 2. Turns out it wasn't a bug... ^^
Anyway, just for completeness, here's synclient output:

~$ synclient 
Parameter settings:
    LeftEdge                = 111
    RightEdge               = 2683
    TopEdge                 = 94
    BottomEdge              = 1664
    FingerLow               = 12
    FingerHigh              = 15
    MaxTapTime              = 180
    MaxTapMove              = 145
    MaxDoubleTapTime        = 100
    SingleTapTimeout        = 180
    ClickTime               = 100
    EmulateMidButtonTime    = 0
    EmulateTwoFingerMinZ    = 142
    EmulateTwoFingerMinW    = 3
    VertScrollDelta         = -66
    HorizScrollDelta        = -66
    VertEdgeScroll          = 0
    HorizEdgeScroll         = 0
    CornerCoasting          = 0
    VertTwoFingerScroll     = 1
    HorizTwoFingerScroll    = 1
    MinSpeed                = 1
    MaxSpeed                = 1.75
    AccelFactor             = 0.0605877
    TouchpadOff             = 2
    LockedDrags             = 0
    LockedDragTimeout       = 5000
    RTCornerButton          = 2
    RBCornerButton          = 3
    LTCornerButton          = 0
    LBCornerButton          = 0
    TapButton1              = 1
    TapButton2              = 3
    TapButton3              = 0
    ClickFinger1            = 1
    ClickFinger2            = 3
    ClickFinger3            = 0
    CircularScrolling       = 0
    CircScrollDelta         = 0.1
    CircScrollTrigger       = 0
    CircularPad             = 0
    PalmDetect              = 0
    PalmMinWidth            = 5
    PalmMinZ                = 100
    CoastingSpeed           = 20
    CoastingFriction        = 50
    PressureMotionMinZ      = 15
    PressureMotionMaxZ      = 80
    PressureMotionMinFactor = 1
    PressureMotionMaxFactor = 1
    ResolutionDetect        = 1
    GrabEventDevice         = 0
    TapAndDragGesture       = 1
    AreaLeftEdge            = 0
    AreaRightEdge           = 0
    AreaTopEdge             = 0
    AreaBottomEdge          = 0
    HorizHysteresis         = 16
    VertHysteresis          = 16
    ClickPad                = 1
    RightButtonAreaLeft     = 1397
    RightButtonAreaRight    = 0
    RightButtonAreaTop      = 1441
    RightButtonAreaBottom   = 0
    MiddleButtonAreaLeft    = 0
    MiddleButtonAreaRight   = 0
    MiddleButtonAreaTop     = 0
    MiddleButtonAreaBottom  = 0

closed 2016-12-16 by yigals