larixer/hid-asus-dkms #29: Tap to click not working

singhriju, 2016-12-01:
Hi,
I can't thank you enough for this driver. 
I just had a small issue, the `tap to click` and `two-finger-tap for right click` feature is not working. It would be great if this works as it would then solve the problem completely.

vlasenko, 2016-12-01:
Hi,

We have seen a bunch of similar issues, all resolved now, maybe you could check solutions inside and report whether anything is relevant for your case?
https://github.com/vlasenko/hid-asus-dkms/issues?q=is%3Aissue+tap+to+click+is%3Aclosed

redmcg, 2016-12-04:
I'll close this as this problem is usually related to the configuration of the Xorg Input driver - and as @vlasenko has already pointed out - there is already solutions posted.

Please do reopen if there does still appear to be a genuine issue with the TouchPad driver.

closed 2016-12-04 by redmcg