larixer/hid-asus-dkms #30: Kernel Module Signing - Idea?

jasonmendes, 2016-12-09:
I somehow have the understanding that secure boot is needed for some third party drivers - I guess this is for the same reason that they must be signed? 

I thought that that meant I couldn't use VirtualBox, but I was able to find a way to self-sign the module, following this guide here: https://gorka.eguileor.com/vbox-vmware-in-secureboot-linux-2016-update/

I wanted to bring it up here because I thought maybe the same method could be used here. I will give it a try myself but I don't yet follow what all the scripts here do, so if you're motivated and capable, I'd be very grateful.

vlasenko, 2016-12-09:
We need to investigate driver signing and what we can do to help DKMS driver users. I'm not sure when we will have time for this, no promises though, but hints are welcome. Pull requests are welcome even more.

jasonmendes, 2016-12-12:
Thanks for your reply! Ah ok, I didn't notice a difference that this is for a driver and not a module? I'd be glad to submit a PR if I can sort this out. Feel free to close this issue, too.

mirh, 2016-12-17:
Can't you use the normal [PreLoader](http://blog.hansenpartnership.com/uefi-secure-boot/) to add the module hash to UEFI sign store and all?

vlasenko, 2016-12-17:
@mirh I don't see in the link you provided any mentions of UEFI loaders or preloaders, neither mentions about UEFI sign store. I'm not sure what do you mean at all. Anyways pull requests are welcome.

vlasenko, 2016-12-17:
@mirh I suspect you are talking about Linux Foundation PreLoader:
http://www.rodsbooks.com/efi-bootloaders/secureboot.html#preloader

mirh, 2016-12-17:
Yes, I'm talking about that (and I mean, it's not like other preloaders exist), which in turn leads you to what I linked. 

Latest sources are [here](http://git.kernel.org/cgit/linux/kernel/git/jejb/efitools.git) then, for as much afaik the majority of distro images already ship with binaries.

redmcg, 2016-12-20:
It does look like there are a number of alternatives to disabling secure boot out there - none look possible to script though (particular this option). I think the best we can do is note that there are alternatives and leave it to the reader/user to choose the option that best suits them.

Commit 6f0979a0b50eb320962b8adf34d7ad892903b458 closes this issue.

closed 2016-12-20 by redmcg

mirh, 2016-12-20:
> It does look like there are a number of alternatives to disabling secure boot out there - none look possible to script though (particular this option)

There are. 
See Bumblebee-Project/Bumblebee#692 and [this](http://askubuntu.com/a/768310/536967). 
The only thing I'm unsure is the one-time addition of your signature to the keystore, although you might want to check [sbsigntools](https://git.kernel.org/cgit/linux/kernel/git/jejb/sbsigntools.git/)

redmcg, 2016-12-20:
@mirh That modification to the Bumblebee install script is to have DKMS automatically sign the installed module on each kernel upgrade (as a DKMS module needs to be rebuilt with each kernel upgrade).

The generation of the keys and the configuration of your firmware to accept these new keys all have to be done manually and by a person physically at the machine. It can not be scripted. Which is a good thing for secure boot - because if it could be scripted - an attacker would use this avenue to circumvent secure boot.

mirh, 2016-12-20:
Well, I'd already find acceptable what Bumblebee already achieved tbh. 

That aside, [sbkeysync](http://jk.ozlabs.org/docs/sbkeysync-maintaing-uefi-key-databases/) seems quite suspiciously interesting.