larixer/hid-asus-dkms #31: wont work with Asus x206h

VirToReal, 2016-12-09:
This is not really a issue than a information:

Thought the x206h got the same Touchpad, but seems there different again... 
After installing via dkms, installation of the driver was as success, but i couldn't use the touchpad anymore. He failed to move it in some new location because it didnt existed, should have made a copy of the output (perhaps it will bring some information). After reboot the original drivers seems back in place, but after pressing any key in the login screen on the keyboard, the Kbuntu 16.10 goes in shutdown ^^. There is no way around by just reinstalling Kubuntu. My bad, just wanted to try, maybe this information will help someone.

vlasenko, 2016-12-09:
Please attach `dmesg` log after you run `./dkms-add.sh`

VirToReal, 2016-12-09:
This is after running install:

> Creating symlink /var/lib/dkms/asus/1.0/source ->
>                  /usr/src/asus-1.0
> 
> DKMS: add completed.
> 
> Kernel preparation unnecessary for this kernel.  Skipping...
> 
> Building module:
> cleaning build area....
> make KERNELRELEASE=4.8.0-30-generic -C /lib/modules/4.8.0-30-generic/build M=/var/lib/dkms/asus/1.0/build/src hid-asus.ko.....
> cleaning build area....
> 
> DKMS: build completed.
> 
> hid-asus.ko:
> Running module version sanity check.
>  - Original module
>  - Installation
>    - Installing to /lib/modules/4.8.0-30-generic/updates/dkms/
> 
> depmod.....
> 
> Backing up initrd.img-4.8.0-30-generic to /boot/initrd.img-4.8.0-30-generic.old-dkms
> Making new initrd.img-4.8.0-30-generic
> (If next boot fails, revert to initrd.img-4.8.0-30-generic.old-dkms image)
> update-initramfs....................
> 
> DKMS: install completed.
> Rebinding 0018:0B05:0101.0002 to hid-asus
> sh: 1: cannot create /sys/bus/hid/drivers/hid-asus/bind: Directory nonexistent
> hardy@hardy-E200HA:~/hid-asus-dkms$ 

`[    0.000000] Linux version 4.8.0-30-generic (buildd@lcy01-08) (gcc version 6.2.0 20161005 (Ubuntu 6.2.0-5ubuntu12) ) #32-Ubuntu SMP Fri Dec 2 03:43:27 UTC 2016 (Ubuntu 4.8.0-30.32-generic 4.8.6)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.8.0-30-generic.efi.signed root=/dev/mmcblk0p2 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: Legacy x87 FPU detected.
[    0.000000] x86/fpu: Using 'eager' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000008efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000008f000-0x000000000008ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000090000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001effffff] usable
[    0.000000] BIOS-e820: [mem 0x000000001f000000-0x00000000201fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020200000-0x0000000078ff9fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000078ffa000-0x0000000079ffffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007a000000-0x000000007ad98fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ad99000-0x000000007adc8fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007adc9000-0x000000007aff0fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007aff1000-0x000000007b829fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007b82a000-0x000000007b89ffff] type 20
[    0.000000] BIOS-e820: [mem 0x000000007b8a0000-0x000000007b8a5fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007b8a6000-0x000000007b8a6fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007b8a7000-0x000000007b8a9fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007b8aa000-0x000000007b8aafff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007b8ab000-0x000000007bffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000e3ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fea00000-0x00000000feafffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed03000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed06000-0x00000000fed06fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed08000-0x00000000fed09fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1cfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fedbffff] reserved
[... 715 lines cut ...]
[  132.070917] audit: type=1400 audit(1481281049.030:22): apparmor="DENIED" operation="open" profile="/usr/sbin/mysqld-akonadi///usr/sbin/mysqld" name="/etc/xdg/akonadi/mysql-global.conf" pid=1515 comm="mysqld" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[  262.625571] systemd[1]: snapd.refresh.timer: Adding 2h 42min 30.723734s random time.
[  262.625594] systemd[1]: snapd.refresh.timer: Adding 2h 48min 34.260313s random time.
[  275.566541] systemd[1]: snapd.refresh.timer: Adding 4h 44min 11.087675s random time.
[  275.566561] systemd[1]: snapd.refresh.timer: Adding 5h 26min 50.655432s random time.
[  276.259769] systemd[1]: snapd.refresh.timer: Adding 1h 11min 46.528324s random time.
[  276.259791] systemd[1]: snapd.refresh.timer: Adding 6min 42.519669s random time.
[  276.620570] systemd[1]: snapd.refresh.timer: Adding 1h 57min 8.188017s random time.
[  276.620592] systemd[1]: snapd.refresh.timer: Adding 5h 28.700521s random time.
[  278.048411] systemd[1]: snapd.refresh.timer: Adding 4h 9min 54.949654s random time.
[  278.048431] systemd[1]: snapd.refresh.timer: Adding 5h 58min 54.169760s random time.
hardy@hardy-E200HA:~/hid-asus-dkms$ 
`
This lines are made after / while installing:

`[  262.625571] systemd[1]: snapd.refresh.timer: Adding 2h 42min 30.723734s random time.
[  262.625594] systemd[1]: snapd.refresh.timer: Adding 2h 48min 34.260313s random time.
[  275.566541] systemd[1]: snapd.refresh.timer: Adding 4h 44min 11.087675s random time.
[  275.566561] systemd[1]: snapd.refresh.timer: Adding 5h 26min 50.655432s random time.
[  276.259769] systemd[1]: snapd.refresh.timer: Adding 1h 11min 46.528324s random time.
[  276.259791] systemd[1]: snapd.refresh.timer: Adding 6min 42.519669s random time.
[  276.620570] systemd[1]: snapd.refresh.timer: Adding 1h 57min 8.188017s random time.
[  276.620592] systemd[1]: snapd.refresh.timer: Adding 5h 28.700521s random time.
[  278.048411] systemd[1]: snapd.refresh.timer: Adding 4h 9min 54.949654s random time.
[  278.048431] systemd[1]: snapd.refresh.timer: Adding 5h 58min 54.169760s random time.`

vlasenko, 2016-12-09:
Okay, after you run `./dkms-add.sh` new driver wasn't loaded but device was unbind from the original driver though. That's why TouchPad stopped working. I think if you will run `./dev-restore.sh` you will be able to bind back TouchPad to original driver.

Now, I'm curious whether new driver is working if you force it to bind to device temporary, for this to check you need to run `./dev-run.sh`. Could you do it, e.g. run `./dev-run.sh`, check whether touchpad is working and post output of `dmesg` here. And I believe if it won't work you will be able to rebind touchpad to original driver using `./dev-restore.sh`

VirToReal, 2016-12-09:
Amazing, its working after dev-run it vlasenko, just the keyboard seems to press shutdown on each key ^^. The settings of Ubuntu recognizes a multitouch-pad. After restoring the driver, the keyboard is working again, and the touchpad, of course, not.

Here the whole dmesg output:

[    0.000000] Linux version 4.8.0-30-generic (buildd@lcy01-08) (gcc version 6.2.0 20161005 (Ubuntu 6.2.0-5ubuntu12) ) #32-Ubuntu SMP Fri Dec 2 03:43:27 UTC 2016 (Ubuntu 4.8.0-30.32-generic 4.8.6)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.8.0-30-generic.efi.signed root=/dev/mmcblk0p2 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: Legacy x87 FPU detected.
[    0.000000] x86/fpu: Using 'eager' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000008efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000008f000-0x000000000008ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000090000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001effffff] usable
[    0.000000] BIOS-e820: [mem 0x000000001f000000-0x00000000201fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020200000-0x0000000078ff9fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000078ffa000-0x0000000079ffffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007a000000-0x000000007ad98fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ad99000-0x000000007adc8fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007adc9000-0x000000007aff0fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007aff1000-0x000000007b829fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007b82a000-0x000000007b89ffff] type 20
[    0.000000] BIOS-e820: [mem 0x000000007b8a0000-0x000000007b8a5fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007b8a6000-0x000000007b8a6fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007b8a7000-0x000000007b8a9fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007b8aa000-0x000000007b8aafff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007b8ab000-0x000000007bffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000e3ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fea00000-0x00000000feafffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed03000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed06000-0x00000000fed06fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed08000-0x00000000fed09fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1cfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fedbffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffa00000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by American Megatrends
[    0.000000] efi:  ESRT=0x7b828098  ACPI=0x79355000  ACPI 2.0=0x79355000  SMBIOS=0x7b121000  SMBIOS 3.0=0x7b120000 
[    0.000000] esrt: Reserving ESRT space from 0x000000007b828098 to 0x000000007b8280d0.
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x7c000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000000000 mask F80000000 write-back
[    0.000000]   1 base 07E000000 mask FFE000000 uncachable
[    0.000000]   2 base 07D000000 mask FFF000000 uncachable
[    0.000000]   3 base 07C800000 mask FFF800000 uncachable
[    0.000000]   4 base 07C400000 mask FFFC00000 uncachable
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT  
[    0.000000] total RAM covered: 1988M
[    0.000000] Found optimal setting for mtrr clean up
[    0.000000]  gran_size: 64K  chunk_size: 64M         num_reg: 5      lose cover RAM: 0G
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] Base memory trampoline at [ffff8c9ec0098000] 98000 size 24576
[    0.000000] BRK [0x25233000, 0x25233fff] PGTABLE
[... 669 lines cut ...]
[   20.159296] wlp1s0: authenticate with 10:6f:3f:01:50:0d
[   20.197592] wlp1s0: send auth to 10:6f:3f:01:50:0d (try 1/3)
[   20.199776] wlp1s0: authenticated
[   20.204696] wlp1s0: associate with 10:6f:3f:01:50:0d (try 1/3)
[   20.209780] wlp1s0: RX AssocResp from 10:6f:3f:01:50:0d (capab=0x411 status=0 aid=3)
[   20.212487] wlp1s0: associated
[   20.212671] IPv6: ADDRCONF(NETDEV_CHANGE): wlp1s0: link becomes ready
[   27.796141] Bluetooth: RFCOMM TTY layer initialized
[   27.796162] Bluetooth: RFCOMM socket layer initialized
[   27.796176] Bluetooth: RFCOMM ver 1.11
[   29.930617] audit_printk_skb: 27 callbacks suppressed
[... cut ...]
[  260.844582] hid-asus 0018:0B05:8585.0001: input,hidraw0: I2C HID v1.00 Keyboard [PDEC3393:00 0B05:8585] on i2c-PDEC3393:00
[  260.844588] hid-asus 0018:0B05:8585.0001: Asus input not registered
[  260.908064] hid-asus: probe of 0018:0B05:8585.0001 failed with error -12
[  261.055680] input: FTE1000:00 0B05:0101 as /devices/pci0000:00/808622C1:02/i2c-10/i2c-FTE1000:00/0018:0B05:0101.0002/input/input11
[  261.056089] hid-asus 0018:0B05:0101.0002: input,hidraw0: I2C HID v1.00 Mouse [FTE1000:00 0B05:0101] on i2c-FTE1000:00

redmcg, 2016-12-10:
Oops - it looks like I  broke the keyboard driver. I've just made a commit which I think will fix it. Can you please do a `git pull` and try again. Both keyboard and touchpad should work.

VirToReal, 2016-12-10:
Works flawlessly, awesome work you two, thanks a lot!

redmcg, 2016-12-10:
Thanks for reporting the issue and thanks for confirming the fix

closed 2016-12-10 by redmcg