larixer/hid-asus-dkms #32: No such device?

ds0nt, 2016-12-11:
I'm using my ZenBook Pro and I'm getting this error when running the script.

It seems like it's not finding the pattern that is being grepped from dmesg.

Any hints on how to debug this?

Also, is this driver for the touchpad? Silly question, but I'm not actually sure if it's for the touchpad (which doesn't work), or for the touch display (which works.. sortof). Either way I was linkd here from here: https://wiki.archlinux.org/index.php/ASUS_Zenbook_Pro_UX501

My 
```bash
./dkms-add.sh 
[sudo] password for ds0nt: 

-------- Uninstall Beginning --------
Module:  asus
Version: 1.0
Kernel:  4.4.36-1-MANJARO (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

hid-asus.ko:
 - Uninstallation
   - Deleting from: /usr/lib/modules/4.4.36-1-MANJARO/updates/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.

------------------------------
Deleting module version: 1.0
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/asus/1.0/source ->
                 /usr/src/asus-1.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=4.4.36-1-MANJARO -C /usr/lib/modules/4.4.36-1-MANJARO/build M=/var/lib/dkms/asus/1.0/build/src hid-asus.ko...
cleaning build area...
Kernel cleanup unnecessary for this kernel.  Skipping...

DKMS: build completed.

hid-asus.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/4.4.36-1-MANJARO/updates/

depmod...

DKMS: install completed.
Rebinding to hid-asus
sh: line 0: echo: write error: No such device
sh: line 0: echo: write error: No such device

```

redmcg, 2016-12-11:
The driver is for the Touchpad. I think Asus have used different models of Touchpad in their same model of laptop - so it might be your laptop doesn't have the Focaltech Touchpad that this driver is built for.

The easiest way to check is to run:
`xinput`

This will output all your input devices.

If it is Focaltech you'll see either:
- `FTE1001:00 0B05:0101` (when our driver is not running); or
- `Asus TouchPad` (when it is)

I know one of the other users who tried our driver had an Elantech Touchpad. So you can also try:
```
basename `ls -d /sys/bus/acpi/devices/{FT*,EL*} 2> /dev/null`
```
to see if either a `FTE` entry (for Focaltech) or a `ELAN` entry comes up.

ds0nt, 2016-12-11:
I did have the ELAN touchpad. Yeah. Thanks!

closed 2016-12-11 by redmcg