larixer/hid-asus-dkms #33: (Asus FL5900U) Find this error when running dkms-add.sh, sh: echo: I/O error

fengxueem, 2016-12-14:
Creating symlink /var/lib/dkms/asus/1.0/source ->
                 /usr/src/asus-1.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make KERNELRELEASE=4.4.0-53-generic -C /lib/modules/4.4.0-53-generic/build M=/var/lib/dkms/asus/1.0/build/src hid-asus.ko....
cleaning build area....

DKMS: build completed.

hid-asus.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.4.0-53-generic/updates/dkms/

depmod....

Backing up initrd.img-4.4.0-53-generic to /boot/initrd.img-4.4.0-53-generic.old-dkms
Making new initrd.img-4.4.0-53-generic
(If next boot fails, revert to initrd.img-4.4.0-53-generic.old-dkms image)
update-initramfs......

DKMS: install completed.
Rebinding to hid-asus
sh: echo: I/O error
sh: echo: I/O error

fengxueem, 2016-12-14:
Here is what I get from xinput:
luozheren@luozheren-X556UB:~$ xinput --list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ PIXART USB OPTICAL MOUSE                	id=10	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ USB Camera                              	id=11	[slave  keyboard (3)]
    ↳ Asus WMI hotkeys                        	id=12	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=13	[slave  keyboard (3)]

fengxueem, 2016-12-14:
And this is from dmesg:
can't find device of ID00e5
[  184.496390] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  184.496410] pcieport 0000:00:1c.5: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e5(Receiver ID)
[  184.496413] pcieport 0000:00:1c.5:   device [8086:9d15] error status/mask=00000001/00002000
[  184.496415] pcieport 0000:00:1c.5:    [ 0] Receiver Error         (First)

fengxueem, 2016-12-14:
By the way, I read everything from https://bugzilla.kernel.org/show_bug.cgi?id=120181, the action from community really surprises me!

redmcg, 2016-12-14:
Could you please try:
`dev-run.sh`
and also post the output of:
`dmesg | grep hid`

Thanks.

fengxueem, 2016-12-15:
**# Output of "dev-run.sh":**
luozheren@luozheren-X556UB:~/hid-asus-dkms$ ./dev-run.sh 
make: Entering directory '/usr/src/linux-headers-4.4.0-53-generic'
make: Leaving directory '/usr/src/linux-headers-4.4.0-53-generic'
make: Entering directory '/usr/src/linux-headers-4.4.0-53-generic'
  CC [M]  /home/luozheren/hid-asus-dkms/src/hid-asus.o
  MODPOST 1 modules
  CC      /home/luozheren/hid-asus-dkms/src/hid-asus.mod.o
  LD [M]  /home/luozheren/hid-asus-dkms/src/hid-asus.ko
make: Leaving directory '/usr/src/linux-headers-4.4.0-53-generic'
[sudo] luozheren 的密码： 
rmmod: ERROR: Module hid_asus is not currently loaded
Rebinding to hid-asus
sh: echo: I/O error
sh: echo: I/O error

**# Output of "dmesg | grep hid":** 
luozheren@luozheren-X556UB:~/hid-asus-dkms$ dmesg | grep hid
luozheren@luozheren-X556UB:~/hid-asus-dkms$ 
I think nothing about "hid" appears in my dmesg.

redmcg, 2016-12-15:
This line:
```
rmmod: ERROR: Module hid_asus is not currently loaded
```
tells me that the module isn't being loaded. Normally `dmesg` would show an error though. Can you maybe try `dmesg | tail` after running `./dev-run.sh`.

The only reason I've seen so far for the module not loading is if you're running secure boot. Can you confirm you've disabled secure boot?

fengxueem, 2016-12-17:
I checked the BIOS again and the Secure Boot was disabled from there. This is the picture of the BIOS page.[http://m0.mail.sina.com.cn/classic/base_download_att.php?get_cmd=show&file_name=IMG_20161217_114207612.jpg&file_size=1929460.5&mid=044865BB63F9120C9C32B8DABA9A075BAE00000000000001&content_type=image/jpeg&attinfo=4-2284-2574898-base64&fid=new&token=987ad8a3851218691666f9045e21de658b4dd36](url)
And that's what looks like after running `./dev-run.sh` in `dmesg | tail` :
`luozheren@luozheren-X556UB:~/hid-asus-dkms$ dmesg | tail
[   76.295671] pcieport 0000:00:1c.5: AER: Multiple Corrected error received: id=00e5
[   76.296074] pcieport 0000:00:1c.5: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e5(Receiver ID)
[   76.296078] pcieport 0000:00:1c.5:   device [8086:9d15] error status/mask=00000001/00000000
[   76.296081] pcieport 0000:00:1c.5:    [ 0] Receiver Error         (First)
[   76.296091] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[   76.296493] pcieport 0000:00:1c.5: can't find device of ID00e5
[   76.296510] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[   76.296517] pcieport 0000:00:1c.5: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e5(Receiver ID)
[   76.296519] pcieport 0000:00:1c.5:   device [8086:9d15] error status/mask=00000001/00000000
[   76.296522] pcieport 0000:00:1c.5:    [ 0] Receiver Error         (First)
`

vlasenko, 2016-12-17:
@fengxueem Please run `./dev-run.sh` and then post here full output of `dmesg`

fengxueem, 2016-12-21:
After run 'dev-run.sh', I store the ouput of dmesg to a file by this:
`luozheren@luozheren-X556UB:~/hid-asus-dkms$ dmesg >> output.txt`
And the results are followed:
[  212.596324] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  212.596348] pcieport 0000:00:1c.5: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e5(Receiver ID)
[  212.596353] pcieport 0000:00:1c.5:   device [8086:9d15] error status/mask=00000001/00002000
[  212.596358] pcieport 0000:00:1c.5:    [ 0] Receiver Error         (First)
[  212.596367] pcieport 0000:00:1c.5: AER: Multiple Corrected error received: id=00e5
[  212.596720] pcieport 0000:00:1c.5: can't find device of ID00e5
[  212.596725] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  212.596748] pcieport 0000:00:1c.5: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e5(Receiver ID)
[  212.596754] pcieport 0000:00:1c.5:   device [8086:9d15] error status/mask=00000001/00002000
[  212.596758] pcieport 0000:00:1c.5:    [ 0] Receiver Error         (First)
[  212.596766] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  212.597118] pcieport 0000:00:1c.5: can't find device of ID00e5
[  212.597123] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  212.597147] pcieport 0000:00:1c.5: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e5(Receiver ID)
[  212.597153] pcieport 0000:00:1c.5:   device [8086:9d15] error status/mask=00000001/00002000
[  212.597156] pcieport 0000:00:1c.5:    [ 0] Receiver Error         (First)
[  212.597163] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  212.597545] pcieport 0000:00:1c.5: can't find device of ID00e5
[  212.597552] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  212.597576] pcieport 0000:00:1c.5: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e5(Receiver ID)
[  212.597588] pcieport 0000:00:1c.5:   device [8086:9d15] error status/mask=00000001/00002000
[  212.597600] pcieport 0000:00:1c.5:    [ 0] Receiver Error         (First)
[  212.597616] pcieport 0000:00:1c.5: AER: Multiple Corrected error received: id=00e5
[  212.597978] pcieport 0000:00:1c.5: can't find device of ID00e5
[  212.597998] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  212.598009] pcieport 0000:00:1c.5: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e5(Receiver ID)
[  212.598014] pcieport 0000:00:1c.5:   device [8086:9d15] error status/mask=00000001/00002000
[  212.598018] pcieport 0000:00:1c.5:    [ 0] Receiver Error         (First)
[  212.598027] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  212.598380] pcieport 0000:00:1c.5: can't find device of ID00e5
[  212.598401] pcieport 0000:00:1c.5: AER: Multiple Corrected error received: id=00e5
[  212.598760] pcieport 0000:00:1c.5: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e5(Receiver ID)
[  212.598779] pcieport 0000:00:1c.5:   device [8086:9d15] error status/mask=00000001/00002000
[  212.598784] pcieport 0000:00:1c.5:    [ 0] Receiver Error         (First)
[  212.598798] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  212.599152] pcieport 0000:00:1c.5: can't find device of ID00e5
[  212.599171] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  212.599184] pcieport 0000:00:1c.5: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e5(Receiver ID)
[  212.599190] pcieport 0000:00:1c.5:   device [8086:9d15] error status/mask=00000001/00002000
[  212.599195] pcieport 0000:00:1c.5:    [ 0] Receiver Error         (First)
[  212.599204] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  212.599561] pcieport 0000:00:1c.5: can't find device of ID00e5
[  212.599579] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  212.599590] pcieport 0000:00:1c.5: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e5(Receiver ID)
[  212.599595] pcieport 0000:00:1c.5:   device [8086:9d15] error status/mask=00000001/00002000
[  212.599599] pcieport 0000:00:1c.5:    [ 0] Receiver Error         (First)
[  212.599608] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  212.599967] pcieport 0000:00:1c.5: can't find device of ID00e5
[  212.599972] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  212.599996] pcieport 0000:00:1c.5: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e5(Receiver ID)
[  212.600001] pcieport 0000:00:1c.5:   device [8086:9d15] error status/mask=00000001/00002000
[  212.600006] pcieport 0000:00:1c.5:    [ 0] Receiver Error         (First)
[  212.600015] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  212.600368] pcieport 0000:00:1c.5: can't find device of ID00e5
[  212.600373] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  212.600397] pcieport 0000:00:1c.5: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e5(Receiver ID)
[  212.600403] pcieport 0000:00:1c.5:   device [8086:9d15] error status/mask=00000001/00002000
[  212.600408] pcieport 0000:00:1c.5:    [ 0] Receiver Error         (First)
[  212.600416] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  212.600769] pcieport 0000:00:1c.5: can't find device of ID00e5
[  212.600774] pcieport 0000:00:1c.5: AER: Corrected error received: id=00e5
[  212.600798] pcieport 0000:00:1c.5: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e5(Receiver ID)
[... cut ...]
[  216.073936] pcieport 0000:00:1c.5:   device [8086:9d15] error status/mask=00000001/00002000
[  216.073947] pcieport 0000:00:1c.5:    [ 0] Receiver Error        

So, it's just repeating of 
```
pcieport 0000:00:1c.5: AER: Multiple Corrected error received: id=00e5
pcieport 0000:00:1c.5: can't find device of ID00e5
pcieport 0000:00:1c.5: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=00e5(Receiver ID)
pcieport 0000:00:1c.5:   device [8086:9d15] error status/mask=00000001/00002000
pcieport 0000:00:1c.5:    [ 0] Receiver Error        
```

vlasenko, 2016-12-21:
@fengxueem So, this is the problem, your dmesg is completely uninformative at the moment. We need to fix this to move further.

Seems you are affected by this problem
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1521173

Please try workaround posted there `pci=noaer`, then boot, run `./dev-run.sh` and post full dmesg here if it will not be all spammed with AER errors.

fengxueem, 2016-12-23:
Thx! @vlasenko . This "pci=noaer" does work! And the new output :
```
luozheren@luozheren-X556UB:~/hid-asus-dkms$ dmesg 
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.0-57-generic (buildd@lgw01-54) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4) ) #78-Ubuntu SMP Fri Dec 9 23:50:32 UTC 2016 (Ubuntu 4.4.0-57.78-generic 4.4.35)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.4.0-57-generic.efi.signed root=UUID=dbdced44-3e83-4662-8f43-70954dc4c6bd ro quiet splash pci=noaer vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  960, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]: 1024, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x04: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x08: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x10: 'MPX CSR'
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 1088 bytes, using 'standard' format.
[    0.000000] x86/fpu: Using 'eager' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000072ebafff] usable
[    0.000000] BIOS-e820: [mem 0x0000000072ebb000-0x0000000072ebbfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000072ebc000-0x0000000072ee5fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000072ee6000-0x00000000755dcfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000755dd000-0x00000000762b9fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000762ba000-0x0000000076592fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000076593000-0x0000000077132fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000077133000-0x0000000077f88fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000077f89000-0x0000000077ffefff] type 20
[    0.000000] BIOS-e820: [mem 0x0000000077fff000-0x0000000077ffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000078000000-0x00000000780fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x0000000381ffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by American Megatrends
[    0.000000] efi:  ESRT=0x77ed0e18  ACPI=0x76697000  ACPI 2.0=0x76697000  SMBIOS=0x776ea000  SMBIOS 3.0=0x776e9000 
[    0.000000] esrt: Reserving ESRT space from 0x0000000077ed0e18 to 0x0000000077ed0e50.
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x382000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: write-back
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 0080000000 mask 7F80000000 uncachable
[    0.000000]   1 base 007C000000 mask 7FFC000000 uncachable
[    0.000000]   2 base 007A000000 mask 7FFE000000 uncachable
[    0.000000]   3 base 0079000000 mask 7FFF000000 uncachable
[    0.000000]   4 base 0078800000 mask 7FFF800000 uncachable
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000]   8 disabled
[    0.000000]   9 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT  
[    0.000000] e820: last_pfn = 0x78000 max_arch_pfn = 0x400000000
[... 877 lines cut ...]
[    9.332908] wlp3s0: authenticated
[    9.333956] wlp3s0: associate with ec:17:2f:22:14:ca (try 1/3)
[    9.343862] wlp3s0: RX AssocResp from ec:17:2f:22:14:ca (capab=0x431 status=0 aid=1)
[    9.344408] wlp3s0: associated
[    9.344417] IPv6: ADDRCONF(NETDEV_CHANGE): wlp3s0: link becomes ready
[    9.346538] cfg80211: Regulatory domain changed to country: US
[    9.346540] cfg80211:  DFS Master region: FCC
[    9.346540] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[    9.346542] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 3000 mBm), (N/A)
[    9.346543] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 1700 mBm), (N/A)
[    9.346545] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2300 mBm), (0 s)
[... cut ...]
[    9.396359] wlp3s0: Limiting TX power to 20 (20 - 0) dBm as advertised by ec:17:2f:22:14:ca
[   10.449888] Bluetooth: RFCOMM TTY layer initialized
[   10.449893] Bluetooth: RFCOMM socket layer initialized
[   10.449896] Bluetooth: RFCOMM ver 1.11
[   13.590084] i2c_hid i2c-FTE1001:00: failed to reset device.
[   19.594460] i2c_hid i2c-FTE1001:00: failed to reset device.
[   25.602329] i2c_hid i2c-FTE1001:00: failed to reset device.
[   26.606819] i2c_hid i2c-FTE1001:00: can't add hid device: -61
[   26.622827] i2c_hid: probe of i2c-FTE1001:00 failed with error -61

```

redmcg, 2016-12-23:
The 'failed to reset device' that you can see at the bottom of your dmesg is discussed in issue #24. The reset takes place within the i2c-hid module. I've made a change to this module in [my fork](https://github.com/redmcg/hid-asus-dkms) of this repo. Could you please try using this and see if it resolves your problem?

fengxueem, 2016-12-25:
I tried this fork and failed. This is the output of `dmesg | grep hid` after running `dev-run.sh`:
```
luozheren@luozheren-X556UB:~/下载/hid-asus-dkms-master$ dmesg | grep hid
[    1.110906] hidraw: raw HID events driver (C) Jiri Kosina
[    1.110973] i2c_hid: module verification failed: signature and/or required key missing - tainting kernel
[    3.602626] usbcore: registered new interface driver usbhid
[    3.602628] usbhid: USB HID core driver
[    3.609294] hid-generic 0003:0461:4D81.0001: input,hidraw0: USB HID v1.11 Mouse [USB Optical Mouse] on usb-0000:00:14.0-2/input0
[    7.589777] i2c_hid i2c-FTE1001:00: failed to reset device.
[   13.597383] i2c_hid i2c-FTE1001:00: failed to reset device.
[   19.609512] i2c_hid i2c-FTE1001:00: failed to reset device.
[   25.621593] i2c_hid i2c-FTE1001:00: failed to reset device.
[   26.625855] i2c_hid i2c-FTE1001:00: can't add hid device: -61
[   26.645628] i2c_hid: probe of i2c-FTE1001:00 failed with error -61
```
And the full output of `dmesg` goes like this:
```
luozheren@luozheren-X556UB:~/下载/hid-asus-dkms-master$ dmesg 
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.0-57-generic (buildd@lgw01-54) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4) ) #78-Ubuntu SMP Fri Dec 9 23:50:32 UTC 2016 (Ubuntu 4.4.0-57.78-generic 4.4.35)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.4.0-57-generic.efi.signed root=UUID=dbdced44-3e83-4662-8f43-70954dc4c6bd ro quiet splash pci=noaer vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  960, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]: 1024, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x04: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x08: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x10: 'MPX CSR'
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 1088 bytes, using 'standard' format.
[    0.000000] x86/fpu: Using 'eager' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000072ebafff] usable
[    0.000000] BIOS-e820: [mem 0x0000000072ebb000-0x0000000072ebbfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000072ebc000-0x0000000072ee5fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000072ee6000-0x00000000755dcfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000755dd000-0x00000000762b9fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000762ba000-0x0000000076592fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000076593000-0x0000000077132fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000077133000-0x0000000077f88fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000077f89000-0x0000000077ffefff] type 20
[    0.000000] BIOS-e820: [mem 0x0000000077fff000-0x0000000077ffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000078000000-0x00000000780fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x0000000381ffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by American Megatrends
[    0.000000] efi:  ESRT=0x77ed0e18  ACPI=0x76697000  ACPI 2.0=0x76697000  SMBIOS=0x776ea000  SMBIOS 3.0=0x776e9000 
[    0.000000] esrt: Reserving ESRT space from 0x0000000077ed0e18 to 0x0000000077ed0e50.
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x382000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: write-back
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[... 888 lines cut ...]
[    9.333223] wlp3s0: authenticated
[    9.337297] wlp3s0: associate with f4:83:cd:9c:05:62 (try 1/3)
[    9.344592] wlp3s0: RX AssocResp from f4:83:cd:9c:05:62 (capab=0x431 status=0 aid=6)
[    9.345088] wlp3s0: associated
[    9.345096] IPv6: ADDRCONF(NETDEV_CHANGE): wlp3s0: link becomes ready
[    9.346800] cfg80211: Regulatory domain changed to country: CN
[    9.346802] cfg80211:  DFS Master region: FCC
[    9.346803] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[... cut ...]
[    9.346810] cfg80211:   (63720000 KHz - 65880000 KHz @ 2160000 KHz), (N/A, 2800 mBm), (N/A)
[   10.463141] Bluetooth: RFCOMM TTY layer initialized
[   10.463147] Bluetooth: RFCOMM socket layer initialized
[   10.463152] Bluetooth: RFCOMM ver 1.11
[   13.597383] i2c_hid i2c-FTE1001:00: failed to reset device.
[   19.609512] i2c_hid i2c-FTE1001:00: failed to reset device.
[   25.621593] i2c_hid i2c-FTE1001:00: failed to reset device.
[   26.625855] i2c_hid i2c-FTE1001:00: can't add hid device: -61
[   26.645628] i2c_hid: probe of i2c-FTE1001:00 failed with error -61

```

vlasenko, 2016-12-25:
@fengxueem 
```
[    1.110973] i2c_hid: module verification failed: signature and/or required key missing - tainting kernel
```
This line means, you are using Linux Kernel that requires signed Kernel Modules. Our module is not signed. You need to sign it manually or use Linux Kernel that don't require signed modules...

vlasenko, 2016-12-26:
@fengxueem Sorry, I'm wrong. Looks like the kernel module is loaded. But it is unable to reset touchpad. Seems some problem with reset in the module...

redmcg, 2016-12-28:
-61 is `ENODATA` which is what the `i2c-hid` module returns when it doesn't get a response from the device after sending a RESET command. This is what is discussed in issue #24 and the attempted resolution was to add a delay after sending the SET_POWER(ON) command (prior to the RESET command).

This doesn't seem to be working in your case.

Can you please try creating a file called `/etc/modprobe.d/i2c-debug.conf` and put the following contents in it:
`options i2c_hid debug=1`

You will then need to run `update-initramfs -u` and reboot. This will turn on additional debugging within the `i2c-hid` module and give a bit more insight about the timing (to ensure the added delay is taking place).

Can you also confirm the Touchpad works perfectly from within Windows? I want to rule out a hardware issue.

wangxuanjie, 2016-12-31:
I have a ASUS FL5900 too,the touchpad work perfectly,after disabling the secure boot and installing this driver

vlasenko, 2017-01-02:
Closing because of topic starter inactivity. Please reopen when you have time to further discuss this issue.

closed 2017-01-02 by vlasenko