larixer/hid-asus-dkms #36: (Asus X556UJ) Find this error when running dkms-add.sh, sh: echo: I/O error

botinada, 2016-12-31:
Creating symlink /var/lib/dkms/asus/1.0/source ->
                 /usr/src/asus-1.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make KERNELRELEASE=4.4.0-57-generic -C /lib/modules/4.4.0-57-generic/build M=/var/lib/dkms/asus/1.0/build/src modules....
cleaning build area....

DKMS: build completed.

hid-asus.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.4.0-57-generic/updates/dkms/

i2c-hid.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.4.0-57-generic/updates/dkms/

depmod....

Backing up initrd.img-4.4.0-57-generic to /boot/initrd.img-4.4.0-57-generic.old-dkms
Making new initrd.img-4.4.0-57-generic
(If next boot fails, revert to initrd.img-4.4.0-57-generic.old-dkms image)
update-initramfs......

DKMS: install completed.
Rebinding to hid-asus
sh: echo: I/O error
sh: echo: I/O error

botinada, 2016-12-31:
# xinput --list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ USB Optical Mouse                       	id=14	[slave  pointer  (2)]
⎜   ↳ Elan Touchpad                           	id=11	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ USB Camera                              	id=10	[slave  keyboard (3)]
    ↳ Asus WMI hotkeys                        	id=12	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=13	[slave  keyboard (3)]

botinada, 2016-12-31:
# dmesg
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.0-57-generic (buildd@lgw01-54) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4) ) #78-Ubuntu SMP Fri Dec 9 23:50:32 UTC 2016 (Ubuntu 4.4.0-57.78-generic 4.4.35)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.4.0-57-generic.efi.signed root=UUID=eee6f240-477d-45a0-b235-880524cf0f82 ro elevator=noop quiet splash pci=noaer psmouse.proto=bare vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  960, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]: 1024, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x04: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x08: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x10: 'MPX CSR'
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 1088 bytes, using 'standard' format.
[    0.000000] x86/fpu: Using 'eager' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000072b85fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000072b86000-0x0000000072b86fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000072b87000-0x0000000072bb0fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000072bb1000-0x00000000755d1fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000755d2000-0x00000000762aefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000762af000-0x0000000076587fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000076588000-0x0000000077126fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000077127000-0x0000000077f87fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000077f88000-0x0000000077ffdfff] type 20
[    0.000000] BIOS-e820: [mem 0x0000000077ffe000-0x0000000077ffefff] usable
[    0.000000] BIOS-e820: [mem 0x0000000078000000-0x00000000780fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x0000000281ffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by American Megatrends
[    0.000000] efi:  ESRT=0x77ec7e18  ACPI=0x7668c000  ACPI 2.0=0x7668c000  SMBIOS=0x776e1000  SMBIOS 3.0=0x776e0000 
[    0.000000] esrt: Reserving ESRT space from 0x0000000077ec7e18 to 0x0000000077ec7e50.
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x282000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: write-back
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 0080000000 mask 7F80000000 uncachable
[    0.000000]   1 base 007C000000 mask 7FFC000000 uncachable
[    0.000000]   2 base 007A000000 mask 7FFE000000 uncachable
[    0.000000]   3 base 0079000000 mask 7FFF000000 uncachable
[    0.000000]   4 base 0078800000 mask 7FFF800000 uncachable
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000]   8 disabled
[    0.000000]   9 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT  
[    0.000000] e820: last_pfn = 0x77fff max_arch_pfn = 0x400000000
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] Base memory trampoline at [ffff880000098000] 98000 size 24576
[... 952 lines cut ...]
[    6.183646] vgaarb: this pci device is not a vga device
[  115.861844] usb 1-1: new low-speed USB device number 5 using xhci_hcd
[  116.051598] usb 1-1: New USB device found, idVendor=1bcf, idProduct=0007
[  116.051615] usb 1-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[  116.051624] usb 1-1: Product: USB Optical Mouse
[  116.052214] usb 1-1: ep 0x81 - rounding interval to 64 microframes, ep desc says 80 microframes
[  116.150299] usbcore: registered new interface driver usbhid
[  116.150301] usbhid: USB HID core driver
[  116.155340] input: USB Optical Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:1BCF:0007.0001/input/input13
[... cut ...]
[  125.029848] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[  125.029850] cfg80211:   (2400000 KHz - 2483500 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  125.029851] cfg80211:   (5150000 KHz - 5250000 KHz @ 80000 KHz, 200000 KHz AUTO), (N/A, 2301 mBm), (N/A)
[  125.029853] cfg80211:   (5250000 KHz - 5350000 KHz @ 80000 KHz, 200000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[  125.029854] cfg80211:   (5470000 KHz - 5725000 KHz @ 160000 KHz), (N/A, 2698 mBm), (0 s)
[  125.029855] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)

botinada, 2016-12-31:
# dmesg | grep hid
[    1.103829] hidraw: raw HID events driver (C) Jiri Kosina
[    1.103901] i2c_hid: module verification failed: signature and/or required key missing - tainting kernel
[  116.150299] usbcore: registered new interface driver usbhid
[  116.150301] usbhid: USB HID core driver
[  116.155489] hid-generic 0003:1BCF:0007.0001: input,hiddev0,hidraw0: USB HID v1.10 Mouse [USB Optical Mouse] on usb-0000:00:14.0-1/input0

redmcg, 2016-12-31:
@botinada Thanks for raising the issue here. This line:
```
Elan Touchpad id
```
from `xinput` and
```
[ 3.118825] input: Elan Touchpad as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-6/i2c-ELAN1000:00/input/input8
```
from your `dmesg` indicate you have an ElanTech Touchpad. The driver here isn't intended for that model of Touchpad. This device will instead use the `elan_i2c` driver.

You'll need to google your symptoms along with elan_i2c. Hopefully something like [this](http://askubuntu.com/questions/809034/16-04-elantech-touchpad-not-working-on-new-laptop) will help you get started.

I'll close this issue as we can't really help you further. But I do hope you are able to resolve your issue and continue to enjoy using Linux!

closed 2016-12-31 by redmcg