larixer/hid-asus-dkms #39: Tap not always accepted as click action.

ceejayrao, 2017-01-08:
Hi,

In my experience so far, there is one issue I've noticed, which is single finger touch is not always recognized as click action.  when this happen, I just move the mouse pointer and then point it back and touch to issue a click action. Is this related to touchpad touch sensitivity setting or the driver itself?

Thank you,
Jitendra.

vlasenko, 2017-01-09:
Hi Jitendra,

To understand whether this is driver or not, you can use `sudo evtest`. If you will be able to click and don't see `BTN_TOUCH` event, then it's a driver problem, if you will always see `BTN_TOUCH` event then it's not a driver problem.

ceejayrao, 2017-01-10:
Thank you that helped me see the event BT_TOUCH with a value 0 when ABS_MT_PRESSURE value 68. Changed the touchpad settings and it started to work.

Event: time 1484047539.564423, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 17970
Event: time 1484047539.564423, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 1496
Event: time 1484047539.564423, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 890
Event: time 1484047539.564423, type 3 (EV_ABS), code 58 (ABS_MT_PRESSURE), value 63
Event: time 1484047539.564423, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1484047539.564423, type 1 (EV_KEY), code 325 (BTN_TOOL_FINGER), value 1
Event: time 1484047539.564423, type 3 (EV_ABS), code 0 (ABS_X), value 1496
Event: time 1484047539.564423, type 3 (EV_ABS), code 1 (ABS_Y), value 890
Event: time 1484047539.564423, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 63
Event: time 1484047539.564423, -------------- SYN_REPORT ------------
Event: time 1484047539.574110, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1484047539.574110, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1484047539.574110, type 1 (EV_KEY), code 325 (BTN_TOOL_FINGER), value 0
Event: time 1484047539.574110, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 0
Event: time 1484047539.574110, -------------- SYN_REPORT ------------

closed 2017-01-10 by ceejayrao