larixer/hid-asus-dkms #40: Asus ZenBook UX501VW issues

MichalDanielDobrzanski, 2017-01-19:
Dear all,

First of all, thank you very much for this driver.
I have successfully added it into my Ubuntu 16.04 kernel 4.6.0.
What I have noticed is that sometimes cursor teleports to different places on the screen.
In some occasions it performs something which is later interpreted as Alt +Tab command.
Also, scaling feature does not work - however, this is not very important for me.

On my dual-booted Window OS keys (which are on the surface of touchpad) work fine. Here, on Ubuntu, however there is some kind of a **bug**:
When I put a finger on 'right' plane of touchpad, then add finger on a buttons plane and finally remove that first finger, the mouse moves event when I am at the button plane of the touchpad. When I place finger on the button plane first, it works fine.

This is quite annoying during regular use.

Is there any way to get this fixed? Or is there any command to view and modify driver settings?

Regards,
Michał Dobrzański

redmcg, 2017-01-19:
Hi Michał,

Thanks for testing this driver and reporting back your findings.

It sounds like you are describing four separate issues so I will number them and address each one separately.

1. Random cursor teleport
2. Random alt-tab
3. Scaling feature doesn't work
4. Cursor jumps when using two fingers and removing the finger placed first

I'm not 100% I understand the fourth issue correctly. When you say "Window OS keys" - do you just mean the left and right mouse button at the bottom of the Touchpad? Or are there other buttons/keys?

We did have an issue in a much earlier version of that driver that could produce issue 1 and 4 - so please confirm you are using the latest version of the driver.

For issue 2 - there is actually a feature that allows you to alt-tab using the Touchpad. To perform this function, you need to double tap the Touchpad with three fingers and hold. Eventually the alt-tab panel will come up and you can scroll to select which application you want to take focus.

For issue 3 - again I'm not 100% sure I understand. What is the scaling feature? Do you mean the ability to zoom for certain applications?

As an FYI - the Touchpad driver is only responsible for two basic things:
1. Power management (i.e. provide the kernel with the ability to turn on or off the Touchpad); and
2. Interpret the data coming from the Touchpad and pass it to "userspace" using the "Input" API.

Features like cursor location, scrolling, zooming and alt-tab are all provided by the Xorg framework which interprets data from the "Input" API (provided by the Touchpad driver). So the issues you are describing could be:
1. An issue in the Xorg framework (where it does not interpret the "Input" API correctly);
2. An issue in the Touchpad driver (where it does not interpret the data from the Touchpad correctly); or
3. An issue with the Touchpad (where it does not send the correct data for the physical touches)

We'll need to dig deeper in to each issue to understand where the underlying issue is. We're probably better off starting with issues 3 and 4 given 1 and 2 appear to be random.

For issue 3 - I need a better understanding. So please elaborate on what actions you are performing along with what you expect and what you currently see.

Same with issue 4. But please also run:
`sudo evtest`
then select the number corresponding to the `Asus TouchPad`. 

This will output all the data sent by the Touchpad over the "Input" API. If you can then recreate these issues and provide the output - I should be able to identify if there is an issue with what the driver is sending or with how the Xorg framework interpreted it.

But the first step is to confirm you are using the latest driver version - and to upgrade if you are not.

vlasenko, 2017-01-30:
@MichalDanielDobrzanski I'm closing this issue because of inactivity. Please reopen, when you will have time on your side to respond here.

closed 2017-01-30 by vlasenko