larixer/hid-asus-dkms #41: Install Error

fillipefeitosa, 2017-01-23:
Hello There and Thx a lot for creating this script.

When I run the commands (as root), the script returns the following error to me:

```
root@cyclope-debian:~/hid-asus-dkms# ./dkms-add.sh 
sudo: dkms: command not found
sudo: dkms: command not found
modprobe: FATAL: Module hid_asus not found.
Rebinding to hid-asus
sh: echo: I/O error
sh: 1: cannot create /sys/bus/hid/drivers/hid-asus/bind: Directory nonexistent
```

Running on a Debian 8, Asus K501U

fillipefeitosa, 2017-01-23:
This was my output **after** dkms install:

```
root@cyclope-debian:~/hid-asus-dkms# ./dkms-add.sh 

Creating symlink /var/lib/dkms/asus/1.0/source ->
                 /usr/src/asus-1.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make KERNELRELEASE=3.16.0-4-amd64 -C /lib/modules/3.16.0-4-amd64/build M=/var/lib/dkms/asus/1.0/build/src modules....(bad exit status: 2)
Error! Bad return status for module build on kernel: 3.16.0-4-amd64 (x86_64)
Consult /var/lib/dkms/asus/1.0/build/make.log for more information.
modprobe: FATAL: Module hid_asus not found.
Rebinding to hid-asus
sh: echo: I/O error
sh: 1: cannot create /sys/bus/hid/drivers/hid-asus/bind: Directory nonexistent
```

vlasenko, 2017-01-23:
> `Consult /var/lib/dkms/asus/1.0/build/make.log for more information.`

Please attach that file to this issue report.

fillipefeitosa, 2017-01-25:
[make_log.txt](https://github.com/vlasenko/hid-asus-dkms/files/730895/make_log.txt)

This is my make.log, thank you for your attention.

oldelvet, 2017-01-25:
I have seen this issue too. In my case the `hid-asus` module is available at `/sys/bus/hid/drivers/asus/bind`

I could work around the issue by editing `dev-attach.sh` and changing `hid-asus` to `asus`

I am not sure if this is due to an initial attempt to load the module before udev rules were in place or whether Ubuntu/Xenial has some interaction with udev and systemd that causes it to get loaded in a different location.

`uname -a` reports
`Linux fisher 4.9.0-040900-generic #201612111631 SMP Sun Dec 11 21:33:00 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`

The following is recorded in dmesg output but I have been loading/removing the driver so it may not be relevant to the issue here.
```

[236430.860664] hid_asus: loading out-of-tree module taints kernel.
[236430.860700] hid_asus: module verification failed: signature and/or required key missing - tainting kernel
[236430.861386] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-FTE1001:00/0018:0B05:0101.0002/input/input16
[236430.861477] hid-asus 0018:0B05:0101.0002: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
[236631.744154] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-FTE1001:00/0018:0B05:0101.0002/input/input17
[236631.744576] hid-generic 0018:0B05:0101.0002: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
[236631.904273] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:633 / ret_size:0
[236631.907660] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:131 / ret_size:0
[236631.907795] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-FTE1001:00/0018:0B05:0101.0003/input/input18
[236631.908033] hid-generic 0018:0B05:0101.0003: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
[236631.968244] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-FTE1001:00/0018:0B05:0101.0003/input/input19
[236631.968806] hid-asus 0018:0B05:0101.0003: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
[236709.217836] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-FTE1001:00/0018:0B05:0101.0003/input/input20
[236709.218154] hid-generic 0018:0B05:0101.0003: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
[236709.355013] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:633 / ret_size:0
[236709.358386] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:131 / ret_size:0
[236709.358459] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-FTE1001:00/0018:0B05:0101.0004/input/input21
[236709.358633] hid-generic 0018:0B05:0101.0004: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
[236709.423940] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-FTE1001:00/0018:0B05:0101.0004/input/input22
[236709.424526] hid-asus 0018:0B05:0101.0004: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
```

vlasenko, 2017-01-26:
@fillipefeitosa Please update the source code of this DKMS module and try again:
https://github.com/vlasenko/hid-asus-dkms#user-content-updating-to-the-latest-driver-version

vlasenko, 2017-01-26:
@oldelvet Would be nice if you copy paste your text into separate issue. Your issue is not related to this one, but it still a problem in this DKMS module and should be fixed.

fillipefeitosa, 2017-01-30:
@vlasenko 
It worked!

my touchpad still not working though. But thanks for your time and the script!

vlasenko, 2017-01-30:
@fillipefeitosa Please clarify, what do you mean by "touchpad still not working" and please paste the output of `dmesg | grep hid` here. Thanks

vlasenko, 2017-01-31:
Closing due to inactivity. Please reopen when you will have time to respond.

closed 2017-01-31 by vlasenko