larixer/hid-asus-dkms #44: Driver binds at /sys/bus/hid/drivers/asus instead of hid-asus

oldelvet, 2017-02-02:
Copied from issue#41 as requested.

On my system the driver initially binded at `/sys/bus/hid/drivers/asus` instead of `/sys/bus/hid/drivers/hid-asus`.

I could work around the issue by editing `dev-attach.sh` and changing `hid-asus` to `asus`

I am not sure if this is due to an initial attempt to load the module before udev rules were in place or whether Ubuntu/Xenial has some interaction with `udev` and `systemd` that causes it to get loaded in a different location.

After a reboot the device is not attached at `hid-asus` as expecting so that suggests that `systemd` may not have been using the new rules initially.

```
uname -a reports
Linux fisher 4.9.0-040900-generic #201612111631 SMP Sun Dec 11 21:33:00 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```

The following is recorded in `dmesg` output but I have been loading/removing the driver so it may not be relevant to the issue here.

```
[236430.860664] hid_asus: loading out-of-tree module taints kernel.
[236430.860700] hid_asus: module verification failed: signature and/or required key missing - tainting kernel
[236430.861386] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-FTE1001:00/0018:0B05:0101.0002/input/input16
[236430.861477] hid-asus 0018:0B05:0101.0002: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
[236631.744154] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-FTE1001:00/0018:0B05:0101.0002/input/input17
[236631.744576] hid-generic 0018:0B05:0101.0002: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
[236631.904273] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:633 / ret_size:0
[236631.907660] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:131 / ret_size:0
[236631.907795] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-FTE1001:00/0018:0B05:0101.0003/input/input18
[236631.908033] hid-generic 0018:0B05:0101.0003: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
[236631.968244] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-FTE1001:00/0018:0B05:0101.0003/input/input19
[236631.968806] hid-asus 0018:0B05:0101.0003: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
[236709.217836] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-FTE1001:00/0018:0B05:0101.0003/input/input20
[236709.218154] hid-generic 0018:0B05:0101.0003: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
[236709.355013] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:633 / ret_size:0
[236709.358386] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:131 / ret_size:0
[236709.358459] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-FTE1001:00/0018:0B05:0101.0004/input/input21
[236709.358633] hid-generic 0018:0B05:0101.0004: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
[236709.423940] input: FTE1001:00 0B05:0101 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-FTE1001:00/0018:0B05:0101.0004/input/input22
[236709.424526] hid-asus 0018:0B05:0101.0004: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 
```

redmcg, 2017-02-07:
@oldelvet Thanks for the report.

> I could work around the issue by editing `dev-attach.sh` and changing `hid-asus` to `asus`

`hid-asus` is the hid_driver name used in our DKMS module. `asus` is the hid_driver name used in the kernel (since 4.8) for Keyboard support (we added the Touchpad support in 4.10). So if you're seeing `asus` then it suggests the kernel version of this module is being loaded. If that is the case then your Keyboard is probably using the asus driver and maybe the two are tripping over each other.

Unfortunately it's not clear from the logs provided as to what is happening. Are you able to revert the scripts to their original content and reboot. If you can then send the contents of the following commands:
```
dmesg | grep hid
lsmod | grep hid
ls /sys/bus/hid/drivers/{hid-,}asus
cat /sys/bus/hid/drivers/{hid-,}asus/*/input/*/name
```

 I'm hoping it'll be a bit clearer on what is happening.

oldelvet, 2017-02-13:
@redmcg I have tried removing the module using `dev-restore.sh` and `dkms-remove.sh` followed by a reboot. Sadly it seems that the driver attached itself in the expected manner. I have included the output requested anyway but at the moment it seems that the problem is not reproducible.

After module removed and reboot

```
richm@fisher:~$ dmesg | grep hid
[    1.608942] hidraw: raw HID events driver (C) Jiri Kosina
[    3.337393] usbcore: registered new interface driver usbhid
[    3.337393] usbhid: USB HID core driver
[   14.234216] hid-multitouch 0003:0457:11D1.0001: input,hiddev0,hidraw0: USB HID v1.11 Device [USBest Technology SiS HID Touch Controller] on usb-0000:00:14.0-12/input0
[   18.661248] i2c_hid i2c-FTE1001:00: failed to reset device.
[   19.710257] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:633 / ret_size:0
[   19.713616] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:131 / ret_size:0
[   19.714079] hid-generic 0018:0B05:0101.0002: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
richm@fisher:~$ lsmod | grep hid
hid_generic            16384  0
hid_multitouch         20480  0
mac_hid                16384  0
usbhid                 53248  0
i2c_hid                20480  0
hid                   122880  4 i2c_hid,hid_generic,usbhid,hid_multitouch
richm@fisher:~$ ls /sys/bus/hid/drivers/{hid-,}asus
ls: cannot access '/sys/bus/hid/drivers/hid-asus': No such file or directory
ls: cannot access '/sys/bus/hid/drivers/asus': No such file or directory
richm@fisher:~$ cat /sys/bus/hid/drivers/{hid-,}asus/*/input/*/name
cat: '/sys/bus/hid/drivers/hid-asus/*/input/*/name': No such file or directory
cat: '/sys/bus/hid/drivers/asus/*/input/*/name': No such file or directory
richm@fisher:~$ 

richm@fisher:~$ find /lib/modules/ -name "*asus*"
/lib/modules/4.9.0-040900-generic/kernel/drivers/media/rc/keymaps/rc-asus-ps3-100.ko
/lib/modules/4.9.0-040900-generic/kernel/drivers/media/rc/keymaps/rc-asus-pc39.ko
/lib/modules/4.9.0-040900-generic/kernel/drivers/hid/hid-asus.ko
/lib/modules/4.9.0-040900-generic/kernel/drivers/input/tablet/pegasus_notetaker.ko
/lib/modules/4.9.0-040900-generic/kernel/drivers/net/usb/pegasus.ko
/lib/modules/4.9.0-040900-generic/kernel/drivers/platform/x86/asus-wmi.ko
/lib/modules/4.9.0-040900-generic/kernel/drivers/platform/x86/asus-nb-wmi.ko
/lib/modules/4.9.0-040900-generic/kernel/drivers/platform/x86/asus-laptop.ko
/lib/modules/4.9.0-040900-generic/kernel/drivers/platform/x86/asus-wireless.ko
/lib/modules/4.9.0-040900-generic/kernel/drivers/hwmon/asus_atk0110.ko
/lib/modules/4.4.0-62-generic/kernel/drivers/media/rc/keymaps/rc-asus-ps3-100.ko
/lib/modules/4.4.0-62-generic/kernel/drivers/media/rc/keymaps/rc-asus-pc39.ko
/lib/modules/4.4.0-62-generic/kernel/drivers/net/usb/pegasus.ko
/lib/modules/4.4.0-62-generic/kernel/drivers/platform/x86/asus-wmi.ko
/lib/modules/4.4.0-62-generic/kernel/drivers/platform/x86/asus-nb-wmi.ko
/lib/modules/4.4.0-62-generic/kernel/drivers/platform/x86/asus-laptop.ko
/lib/modules/4.4.0-62-generic/kernel/drivers/hwmon/asus_atk0110.ko

```

dkms add

```
richm@fisher:~/Git/hid-asus-dkms$ ./dkms-add.sh 
[sudo] password for richm: 

Creating symlink /var/lib/dkms/asus/1.0/source ->
                 /usr/src/asus-1.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make KERNELRELEASE=4.9.0-040900-generic -C /lib/modules/4.9.0-040900-generic/build M=/var/lib/dkms/asus/1.0/build/src modules....
cleaning build area....

DKMS: build completed.

[... 33 lines cut ...]
[   18.661248] i2c_hid i2c-FTE1001:00: failed to reset device.
[   19.710257] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:633 / ret_size:0
[   19.713616] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:131 / ret_size:0
[   19.714079] hid-generic 0018:0B05:0101.0002: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
[ 1628.507582] hid_asus: loading out-of-tree module taints kernel.
[ 1628.507616] hid_asus: module verification failed: signature and/or required key missing - tainting kernel
[ 1628.589113] hid-asus 0018:0B05:0101.0002: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
richm@fisher:~/Git/hid-asus-dkms$ lsmod | grep hid
hid_asus               16384  0
hid_generic            16384  0
hid_multitouch         20480  0
mac_hid                16384  0
usbhid                 53248  0
i2c_hid                20480  0
hid                   122880  5 i2c_hid,hid_asus,hid_generic,usbhid,hid_multitouch
richm@fisher:~/Git/hid-asus-dkms$ ls /sys/bus/hid/drivers/{hid-,}asus
ls: cannot access '/sys/bus/hid/drivers/asus': No such file or directory
/sys/bus/hid/drivers/hid-asus:
0018:0B05:0101.0002  bind  module  new_id  uevent  unbind
richm@fisher:~/Git/hid-asus-dkms$ cat /sys/bus/hid/drivers/{hid-,}asus/*/input/*/name
Asus TouchPad
cat: '/sys/bus/hid/drivers/asus/*/input/*/name': No such file or directory
richm@fisher:~/Git/hid-asus-dkms$ 

```

redmcg, 2017-02-14:
Are you able to run `dkms-add.sh` and reboot? Then run the same commands:
```
dmesg | grep hid
lsmod | grep hid
ls /sys/bus/hid/drivers/{hid-,}asus
cat /sys/bus/hid/drivers/{hid-,}asus/*/input/*/name
```

I want to make sure the DKMS module is binding properly during your boot.

oldelvet, 2017-02-15:
Yes it is binding fine during reboot. I did a `dkms-remove.sh`, rebooted, then `dkms-add.sh` followed by another reboot and got the following output.
```
richm@fisher:~$ dmesg | grep hid
[    1.605942] hidraw: raw HID events driver (C) Jiri Kosina
[    1.605995] i2c_hid: loading out-of-tree module taints kernel.
[    1.606004] i2c_hid: module verification failed: signature and/or required key missing - tainting kernel
[    2.610339] usbcore: registered new interface driver usbhid
[    2.610340] usbhid: USB HID core driver
[   12.615955] hid-asus 0018:0B05:0101.0002: input,hidraw0: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
[   13.143287] hid-multitouch 0003:0457:11D1.0001: input,hiddev0,hidraw1: USB HID v1.11 Device [USBest Technology SiS HID Touch Controller] on usb-0000:00:14.0-12/input0
richm@fisher:~$ lsmod | grep hid
hid_multitouch         20480  0
hid_generic            16384  0
hid_asus               16384  0
mac_hid                16384  0
usbhid                 53248  0
i2c_hid                20480  0
hid                   122880  5 i2c_hid,hid_asus,hid_generic,usbhid,hid_multitouch
richm@fisher:~$ ls /sys/bus/hid/drivers/{hid-,}asus
ls: cannot access '/sys/bus/hid/drivers/asus': No such file or directory
/sys/bus/hid/drivers/hid-asus:
0018:0B05:0101.0002  bind  module  new_id  uevent  unbind
richm@fisher:~$ cat /sys/bus/hid/drivers/{hid-,}asus/*/input/*/name
Asus TouchPad
cat: '/sys/bus/hid/drivers/asus/*/input/*/name': No such file or directory
richm@fisher:~$ 

```

oldelvet, 2017-02-15:
I have found out what happened and have managed to reproduce the issue.
When I initially tried dkms-add.sh I did not have the dkms package installed so the module was not properly installed and it causes the module shipped with the kernel to install and bind at /sys/bus/hid/drivers/asus. I must have installed dkms without removing the module.

```
richm@fisher:~/Git/hid-asus-dkms$ ./dkms-add.sh 
[sudo] password for richm: 
sudo: dkms: command not found
sudo: dkms: command not found
Rebinding 0018:0B05:0101.0002 to hid-asus
sh: 1: cannot create /sys/bus/hid/drivers/hid-asus/bind: Directory nonexistent
richm@fisher:~/Git/hid-asus-dkms$ dmesg  |grep hid
[    1.607929] hidraw: raw HID events driver (C) Jiri Kosina
[    3.325112] usbcore: registered new interface driver usbhid
[    3.325112] usbhid: USB HID core driver
[   16.065214] hid-multitouch 0003:0457:11D1.0001: input,hiddev0,hidraw0: USB HID v1.11 Device [USBest Technology SiS HID Touch Controller] on usb-0000:00:14.0-12/input0
[   20.449090] i2c_hid i2c-FTE1001:00: failed to reset device.
[   26.593030] i2c_hid i2c-FTE1001:00: failed to reset device.
[   27.643065] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:633 / ret_size:0
[   27.646423] i2c_hid i2c-FTE1001:00: error in i2c_hid_init_report size:131 / ret_size:0
[   27.646754] hid-generic 0018:0B05:0101.0002: input,hidraw1: I2C HID v1.00 Mouse [FTE1001:00 0B05:0101] on i2c-FTE1001:00
richm@fisher:~/Git/hid-asus-dkms$ lsmod | grep hid
hid_asus               16384  0
hid_generic            16384  0
hid_multitouch         20480  0
mac_hid                16384  0
usbhid                 53248  0
i2c_hid                20480  0
hid                   122880  5 i2c_hid,hid_asus,hid_generic,usbhid,hid_multitouch
richm@fisher:~/Git/hid-asus-dkms$ ls /sys/bus/hid/drivers/{hid-,}asus/*/input/*/name
ls: cannot access '/sys/bus/hid/drivers/hid-asus/*/input/*/name': No such file or directory
ls: cannot access '/sys/bus/hid/drivers/asus/*/input/*/name': No such file or directory
richm@fisher:~/Git/hid-asus-dkms$ ls /sys/bus/hid/drivers/{hid-,}asusls: cannot access '/sys/bus/hid/drivers/hid-asus': No such file or directory
/sys/bus/hid/drivers/asus:
bind  module  new_id  uevent  unbind
richm@fisher:~/Git/hid-asus-dkms$ ls /sys/bus/hid/drivers/{hid-,}asus/*/input/*/name
ls: cannot access '/sys/bus/hid/drivers/hid-asus/*/input/*/name': No such file or directory
ls: cannot access '/sys/bus/hid/drivers/asus/*/input/*/name': No such file or directory
richm@fisher:~/Git/hid-asus-dkms$ 

```

I can attach things properly by removing the install, adding dkms and re-adding causes things to get attached properly.

```
./dkms-remove.sh
sudo apt install dkms
./dkms-add.sh

```
I'm not sure what your preferred solution would be but it would be good to add a third prerequisite in the `README.md` to state that the `dkms` package needs to be installed. Maybe also check in `dkms-add.sh` that it is actually installed before proceeding.

vlasenko, 2017-02-15:
@oldelvet I believe testing for `dkms` has been already added with the commit:
https://github.com/vlasenko/hid-asus-dkms/commit/36df3dae97fd1e21bdad7d87a927cbd71d475f71

oldelvet, 2017-02-15:
@vlasenko Many thanks. I have updated from git and can confirm that the dkms check does give a suitable warning and stop the add.

```
richm@fisher:~/Git/hid-asus-dkms$ ./dkms-add.sh 
dkms not found, aborting...

```
That all looks good to me.

vlasenko, 2017-02-15:
@oldelvet Thank you!

closed 2017-02-15 by vlasenko