larixer/hid-asus-dkms #45: dkms-add fails, but attach works

MathieuCliche, 2017-02-04:
I disabled secure boost, but when I run dkms-add I get the following errors : 

sudo: dkms: command not found
sudo: dkms: command not found
modprobe: FATAL: Module hid_asus not found in directory /lib/modules/4.4.0-62-generic
Rebinding 0018:0B05:0101.0003 to hid-asus
sh: echo: I/O error
sh: 1: cannot create /sys/bus/hid/drivers/hid-asus/bind: Directory nonexistent
 
And after that step the track pad doesn't work anymore.  

However, if I run the dev-run, then I still get some errors :  

rmmod: ERROR: Module hid_asus is not currently loaded
modprobe: FATAL: Module hid_asus not found in directory /lib/modules/4.4.0-62-generic
Rebinding 0018:0B05:0101.0004 to hid-asus

But after that the scrolling works fine, at least until reboot.  

Any ideas ?  Thanks!

vlasenko, 2017-02-04:
"sudo: dkms: command not found". You should install `dkms` package on your system first.

MathieuCliche, 2017-02-04:
Oups ! Problem solved, thanks!

closed 2017-02-04 by vlasenko