larixer/hid-asus-dkms #47: do you have a paypal for donations

endzyme, 2017-02-18:


redmcg, 2017-02-20:
@endzyme Hi Nick - thanks for the offer. I don't have a paypal but I was more than happy to volunteer my time to give back to the Linux community. And that's all I'd ask in return. That doesn't have to be writing code/drivers - it can be helping new users on the forum's or converting friends/family to a more productive operating system. Every little bit helps.

I can't speak on behalf of @vlasenko and @ain101 though (who also contributed to the driver). They may just prefer the sweet money cash.

vlasenko, 2017-02-21:
@endzyme @redmcg Hi Nick and Brendan. Nick! I think at the current stage, user base of this project is pretty limited, so I don't see the point to register non-profit Organization, accumulate donations and spread them for development purposes, to much hassle for now. But thank you for your offer! The driver is still not complete, We are still struggling with TouchPad going crazy randomly. Also there are other models of ASUS hardware that is very similar, we can add support for them too.

So I propose to keep this ticket open to let people tell what they want to see in this driver and let them connect to developers and motivate them somehow, if needed.

pabloab, 2017-03-29:
Maybe you could add a [donate in bitcoin](https://github.com/danielquinn/bitcoin-donate) button :-)

aelur, 2017-04-01:
Hey guys, just wanted to drop a quick thank you for all this amazing work you've done. I'm now using/testing this driver, will come back if I find any issues with it. 

Seriously though, massive kudos to you all!!!

closed 2019-02-05 by endzyme