larixer/hid-asus-dkms #48: sh: echo: I/O error running dkms-add.sh

pabloab, 2017-03-29:
At the end of `dkms-add.sh` I get a `sh: echo: I/O error`. The problem seems to be on `dev-attach.sh`, `DEV_NO` is empty because there is no "on i2c-FTE100" on dmesg output. Either way `sudo sh -c` return `sh: echo: I/O error`. Maybe the way would `echo $DEV_NO | sudo tee ...`. With that change I get a an expected `No such device`.

I have an Ubuntu 16.04.2 on Asus UX303UB with 64 bit kernel v4.4.0-70. My FocalTech touchpad was working fine some time ago (I usually use an external mouse). Probably stop working on a kernel upgrade. 

    $ dmesg | grep -iE "touchpad|focaltech|8042"
    i8042: PNP: PS/2 Controller [PNP030b:PS2K,PNP0f03:PS2M] at 0x60,0x64 irq 1,12
    serio: i8042 KBD port at 0x60,0x64 irq 1
    serio: i8042 AUX port at 0x60,0x64 irq 12
    input: AT Translated Set 2 keyboard as /devices/platform/i8042/serio0/input/input3
    input: FocalTechPS/2 FocalTech FocalTech Touchpad as /devices/platform/i8042/serio1/input/input5

    $ dkms status
    asus, 1.0, 4.4.0-70-generic, x86_64: installed

Other things I tried:
- `i8042.reset` and `i8042.noloop=1` on grub options without luck.
- Works again and shows on `xinput` with `sudo modprobe -r psmouse && sudo modprobe psmouse proto=imps && sleep 1 && xinput list` but without multitouch.

vlasenko, 2017-03-29:
@pabloab This driver is for Asus I2C TouchPad. You have different device: FocalTech PS/2 TouchPad.

ain101, 2017-03-29:
@pabloab I see "PS/2" all over the place. This driver was written for i2c. If your touchpad wants to communicate via PS/2 you wont find a solution here.

pabloab, 2017-03-29:
Oops! I found some places (mostly bug reports on launchpad) with almost similar machines pointing here as The solution. Sorry to bother.

closed 2017-03-29 by ain101