larixer/hid-asus-dkms #49: Getting "sh: echo: I/O error" when installing this driver

gauravdhiman, 2017-04-20:
Hi,

First of all thanks for your awesome work to build this driver. Its needed by many.
I am not able to run the driver successfully as per the instruction given on your README page.

When I run `./dkms-add.sh` I get `sh: echo: I/O error` at the end and the trackpad does not work. Looks like the driver gets installed successfully but is not functional or getting some error on initialization.

Below are the logs:
```
gauravdhiman@gd-asus-501:~/proj/hid-asus-dkms$ ./dkms-add.sh
[sudo] password for gauravdhiman: 

-------- Uninstall Beginning --------
Module:  asus
Version: 1.0
Kernel:  4.2.0-42-generic (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

hid-asus.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.2.0-42-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

i2c-hid.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.2.0-42-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod....

Backing up initrd.img-4.2.0-42-generic to /boot/initrd.img-4.2.0-42-generic.old-dkms
Making new initrd.img-4.2.0-42-generic
(If next boot fails, revert to initrd.img-4.2.0-42-generic.old-dkms image)
update-initramfs......

DKMS: uninstall completed.

------------------------------
Deleting module version: 1.0
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/asus/1.0/source ->
                 /usr/src/asus-1.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make KERNELRELEASE=4.2.0-42-generic -C /lib/modules/4.2.0-42-generic/build M=/var/lib/dkms/asus/1.0/build/src modules....
cleaning build area....

DKMS: build completed.

hid-asus.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.2.0-42-generic/updates/dkms/

i2c-hid.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.2.0-42-generic/updates/dkms/

depmod....

Backing up initrd.img-4.2.0-42-generic to /boot/initrd.img-4.2.0-42-generic.old-dkms
Making new initrd.img-4.2.0-42-generic
(If next boot fails, revert to initrd.img-4.2.0-42-generic.old-dkms image)
update-initramfs......

DKMS: install completed.
Rebinding to hid-asus
sh: echo: I/O error
sh: echo: I/O error
```

redmcg, 2017-04-20:
I'd like to confirm this driver is for your hardware. Could you please reboot your laptop and run the following two commands:
```
dmesg | grep hid
xinput
```

and then provide the output? Thanks.

FredySosa, 2017-04-27:
Hi, I'm working on Slackware 14.2, in a ASUS-X556U, and i can't make touchpad works; installing the driver, i have the next log:
```
root@fredy:~/hid-asus-dkms# ./dkms-add.sh
Creating symlink /var/lib/dkms/asus/1.0/source ->
                 /usr/src/asus-1.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.4.14 -C /lib/modules/4.4.14/build M=/var/lib/dkms/asus/1.0/build/src modules...
cleaning build area....

DKMS: build completed.

hid-asus.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.4.14/updates/

i2c-hid.ko:
Running module version sanity check.
 - Original module
   - Found /lib/modules/4.4.14/updates/i2c-hid.ko
   - Storing in /var/lib/dkms/asus/original_module/4.4.14/x86_64/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/4.4.14/updates/

depmod...

: Unable to find an initial ram disk that I know how to handle.
Will not try to make an initrd.

DKMS: install completed.
Rebinding to hid-asus
sh: line 0: echo: write error: No such device
sh: line 0: echo: write error: No such device
root@fredy:~/hid-asus-dkms#
```

It's similar to the output of @gauravdhiman. 
The output of `dmesg | grep hid` and `xinput` is:
```
root@fredy:~# dmesg | grep hid
[    8.419437] hidraw: raw HID events driver (C) Jiri Kosina
[   10.904827] i2c_hid i2c-FTE1001:00: failed to reset device.
[   11.906144] i2c_hid i2c-FTE1001:00: failed to reset device.
[   12.978884] i2c_hid i2c-FTE1001:00: failed to change power setting.
[   12.981265] hid-generic 0018:0B05:0101.0001: item fetching failed at offset 1907783109
[   12.983480] hid-generic: probe of 0018:0B05:0101.0001 failed with error -22
[  450.594090] hid-generic 0003:046D:C51B.0002: input,hidraw0: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-0000:00:14.0-2/input0
[  450.596392] hid-generic 0003:046D:C51B.0003: hiddev0,hidraw1: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:00:14.0-2/input1
[  450.596409] usbcore: registered new interface driver usbhid
[  450.596410] usbhid: USB HID core driver
[  849.085785] hid-asus 0018:0B05:0101.0001: item fetching failed at offset 1571254405
[  849.085800] hid-asus 0018:0B05:0101.0001: Asus hid parse failed: -22
[  849.085810] hid-asus: probe of 0018:0B05:0101.0001 failed with error -22
[ 1027.365070] hid-generic 0003:046D:C51B.0004: input,hidraw0: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-0000:00:14.0-2/input0
[ 1027.368142] hid-generic 0003:046D:C51B.0005: hiddev0,hidraw1: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:00:14.0-2/input1
[ 1047.688770] i2c_hid i2c-FTE1001:00: failed to change power setting.
[ 1048.734867] i2c_hid i2c-FTE1001:00: failed to change power setting.
[ 1049.907942] hid-generic 0018:FFFF:FFFF.0006: item fetching failed at offset 1020821381
[ 1049.907951] hid-generic: probe of 0018:FFFF:FFFF.0006 failed with error -22
[ 1630.819158] i2c_hid i2c-FTE1001:00: unexpected HID descriptor bcdVersion (0x0905)
[ 2750.714739] i2c_hid i2c-FTE1001:00: unexpected HID descriptor bcdVersion (0x0001)
root@fredy:~# 
```
And
```
root@fredy:~# xinput
Virtual core pointer                         id=2    [master pointer  (3)]
    Virtual core XTEST pointer               id=4    [slave  pointer  (2)]
    Logitech USB Receiver                    id=10   [slave  pointer  (2)]
Virtual core keyboard                        id=3    [master keyboard (2)]
    Virtual core XTEST keyboard              id=5    [slave  keyboard (3)]
    Power Button                             id=6    [slave  keyboard (3)]
    Video Bus                                id=7    [slave  keyboard (3)]
    Video Bus                                id=8    [slave  keyboard (3)]
    Sleep Button                             id=9    [slave  keyboard (3)]
    Asus WMI hotkeys                         id=12   [slave  keyboard (3)]
    AT Translated Set 2 keyboard             id=13   [slave  keyboard (3)]
    Speakup                                  id=14   [slave  keyboard (3)]
    USB2.0 VGA UVC WebCam                    id=11   [slave  keyboard (3)]
root@fredy:~# 
```
Where the Logitec USB Reciever is an external mouse.
I'm stuck with that since weeks ago.

redmcg, 2017-04-28:
@FredySosa Thanks for the report.

It looks like the Touchpad isn't talking the HID protocol. Can you please:
a) uninstall our driver, reboot and provide the output of `dmesg | grep hid` and `xinput`; and
b) let me know if the device behaves perfectly in Windows?

With our driver uninstalled - the `hid-generic` driver should enable you to use this device as a mouse. I also want to rule out a hardware issue by making sure the device works correctly in Windows.

FredySosa, 2017-04-28:
Hi @redmcg, answering your questions:
a) The output without the driver is:
```
root@fredy:~# dmesg | grep hid
[    7.714208] hidraw: raw HID events driver (C) Jiri Kosina
[   10.205763] i2c_hid i2c-FTE1001:00: failed to reset device.
[   10.208095] i2c_hid i2c-FTE1001:00: failed to change power setting.
[   11.435554] hid-generic 0018:0B05:0101.0001: item fetching failed at offset 1902902021
[   11.438237] hid-generic: probe of 0018:0B05:0101.0001 failed with error -22
[  159.720161] hid-generic 0003:046D:C51B.0002: input,hidraw0: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-0000:00:14.0-2/input0
[  159.722415] hid-generic 0003:046D:C51B.0003: hiddev0,hidraw1: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:00:14.0-2/input1
[  159.722459] usbcore: registered new interface driver usbhid
[  159.722462] usbhid: USB HID core driver
root@fredy:~# 
```
and
```
root@fredy:~# xinput
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ Logitech USB Receiver                     id=14   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Video Bus                                 id=7    [slave  keyboard (3)]
    ↳ Video Bus                                 id=8    [slave  keyboard (3)]
    ↳ Sleep Button                              id=9    [slave  keyboard (3)]
    ↳ USB2.0 VGA UVC WebCam                     id=10   [slave  keyboard (3)]
    ↳ Asus WMI hotkeys                          id=11   [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard              id=12   [slave  keyboard (3)]
    ↳ Speakup                                   id=13   [slave  keyboard (3)]
root@fredy:~# 
```
b) This pad work fine in Windows, needs and extra controller provided by Asus `ASUS Smart Gesture`, but it's just for the scroll with two fingers and showing the new Windows 10 Desktops with three fingers, but without the Asus Smart Gesture works the movement and clic.
Thank's for the help

vlasenko, 2017-04-28:
@FredySosa The problem the Brendan, aka @redmcg refers is that in order for this driver to work it should be able to communicate over HID protocol with your TouchPad. But `dmesg` logs tell us that Linux kernel cannot establish connection with touchpad over HID protocol. This might be because the TouchPad is behind some I2C BUS which is not supported by Linux kernel yet. So in order for this driver to work Linux kernel should also support that I2C BUS. This is only hypotesis right now... But the driver in this repo relies on working HID protocol to communicate with TouchPad.

vlasenko, 2017-04-28:
@FredySosa Could you also post output of `lspci -v` command here?

FredySosa, 2017-04-28:
Something like that my teacher says: "Maybe it's too new for the linux kernel". Anyway, here's the output @vlasenko 
```
root@fredy:~# lspci -v
00:00.0 Host bridge: Intel Corporation Sky Lake Host Bridge/DRAM Registers (rev 08)
        Subsystem: ASUSTeK Computer Inc. Sky Lake Host Bridge/DRAM Registers
        Flags: bus master, fast devsel, latency 0
        Capabilities: [e0] Vendor Specific Information: Len=10 <?>

00:02.0 VGA compatible controller: Intel Corporation Sky Lake Integrated Graphics (rev 07) (prog-if 00 [VGA controller])
        Subsystem: ASUSTeK Computer Inc. Sky Lake Integrated Graphics
        Flags: bus master, fast devsel, latency 0, IRQ 129
        Memory at dd000000 (64-bit, non-prefetchable) [size=16M]
        Memory at b0000000 (64-bit, prefetchable) [size=256M]
        I/O ports at f000 [size=64]
        Expansion ROM at <unassigned> [disabled]
        Capabilities: [40] Vendor Specific Information: Len=0c <?>
        Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00
        Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable- 64bit-
        Capabilities: [d0] Power Management version 2
        Capabilities: [100] Process Address Space ID (PASID)
        Capabilities: [200] Address Translation Service (ATS)
        Capabilities: [300] Page Request Interface (PRI)
        Kernel driver in use: i915
        Kernel modules: i915

00:04.0 Signal processing controller: Intel Corporation Device 1903 (rev 08)
        Subsystem: ASUSTeK Computer Inc. Device 11c0
        Flags: fast devsel, IRQ 16
        Memory at df320000 (64-bit, non-prefetchable) [size=32K]
        Capabilities: [90] MSI: Enable- Count=1/1 Maskable- 64bit-
        Capabilities: [d0] Power Management version 3
        Capabilities: [e0] Vendor Specific Information: Len=0c <?>
        Kernel driver in use: proc_thermal
        Kernel modules: processor_thermal_device

00:14.0 USB controller: Intel Corporation Device 9d2f (rev 21) (prog-if 30 [XHCI])
        Subsystem: ASUSTeK Computer Inc. Device 201f
        Flags: bus master, medium devsel, latency 0, IRQ 130
        Memory at df310000 (64-bit, non-prefetchable) [size=64K]
        Capabilities: [70] Power Management version 2
        Capabilities: [80] MSI: Enable+ Count=1/8 Maskable- 64bit+
        Kernel driver in use: xhci_hcd
        Kernel modules: xhci_pci

00:14.2 Signal processing controller: Intel Corporation Device 9d31 (rev 21)
        Subsystem: ASUSTeK Computer Inc. Device 11c0
        Flags: bus master, fast devsel, latency 0, IRQ 11
        Memory at df338000 (64-bit, non-prefetchable) [size=4K]
        Capabilities: [50] Power Management version 3
        Capabilities: [80] MSI: Enable- Count=1/1 Maskable- 64bit-

00:15.0 Signal processing controller: Intel Corporation Device 9d60 (rev 21)
        Subsystem: ASUSTeK Computer Inc. Device 11c0
        Flags: bus master, fast devsel, latency 0, IRQ 16
        Memory at df337000 (64-bit, non-prefetchable) [size=4K]
        Capabilities: [80] Power Management version 3
        Capabilities: [90] Vendor Specific Information: Len=14 <?>
        Kernel driver in use: intel-lpss
        Kernel modules: intel_lpss_pci

00:15.1 Signal processing controller: Intel Corporation Device 9d61 (rev 21)
        Subsystem: ASUSTeK Computer Inc. Device 11c0
        Flags: bus master, fast devsel, latency 0, IRQ 17
        Memory at df336000 (64-bit, non-prefetchable) [size=4K]
        Capabilities: [80] Power Management version 3
        Capabilities: [90] Vendor Specific Information: Len=14 <?>
        Kernel driver in use: intel-lpss
        Kernel modules: intel_lpss_pci

00:16.0 Communication controller: Intel Corporation Device 9d3a (rev 21)
[... 124 lines cut ...]
        Capabilities: [140] Virtual Channel
        Capabilities: [160] Device Serial Number 01-00-00-00-68-4c-e0-00
        Capabilities: [170] Latency Tolerance Reporting
        Capabilities: [178] L1 PM Substates
        Kernel driver in use: r8168
        Kernel modules: r8169, r8168

03:00.0 Network controller: Qualcomm Atheros QCA9565 / AR9565 Wireless Network Adapter (rev 01)
        Subsystem: Lite-On Communications Inc QCA9565 / AR9565 Wireless Network Adapter
        Flags: bus master, fast devsel, latency 0, IRQ 17
        Memory at df100000 (64-bit, non-prefetchable) [size=512K]
        Expansion ROM at df180000 [disabled] [size=64K]
        Capabilities: [40] Power Management version 2
        Capabilities: [50] MSI: Enable- Count=1/4 Maskable+ 64bit+
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [140] Virtual Channel
        Capabilities: [160] Device Serial Number 00-00-00-00-00-00-00-00
        Kernel driver in use: ath9k
        Kernel modules: ath9k, wl

root@fredy:~# 
```

Thank's for the help.

redmcg, 2017-04-29:
This could be an I2C communication issue. If you can provide the output of `dmesg | grep i2c` - hopefully we can learn more.

Also - are you able to try a later version of the kernel? I can see you are using v4.4 but in v4.5 a fix was introduced for arbitration in the Skylake LPSS (which hosts the I2C master that talks to your Touchpad):
commit torvalds/linux@028af5941dd870afd5eb6a95c39f25564dcca79a.

The error reportedly fixed in that commit is:
`i2c_dw_handle_tx_abort: lost arbitration`

vlasenko, 2017-06-03:
Closing due to topic starter inactivity

closed 2017-06-03 by vlasenko

gauravdhiman, 2017-11-04:
@vlasenko sorry could not respond earlier.
I tried running it again, but still facing the issue that I reported originally.

Here is the output of `dmesg | grep hid` and `xinput`. Please let me know if you need any other output to make it work. I like Linux but this multi-touch not working on touchpad is disappointing. Will appreciate your help.

```sh
gauravdhiman@gd-asus-501:~$ dmesg | grep hid
[    0.945817] hidraw: raw HID events driver (C) Jiri Kosina
[    1.671674] usbcore: registered new interface driver usbhid
[    1.671676] usbhid: USB HID core driver
[    1.694788]  tpm_crb(+) mac_hid acpi_thermal_rel acpi_pad hid_multitouch(+) usbhid parport_pc ppdev lp parport autofs4 nouveau i915 mxm_wmi ttm i2c_algo_bit drm_kms_helper drm ahci libahci i2c_hid hid wmi pinctrl_sunrisepoint video pinctrl_intel
[    1.695506] hid-multitouch 0003:0457:11D1.0001: input,hiddev0,hidraw0: USB HID v1.11 Device [USBest Technology SiS HID Touch Controller] on usb-0000:00:14.0-12/input0
[    2.178267] i2c_hid i2c-ELAN1000:00: error in i2c_hid_init_report size:653 / ret_size:7
[    2.182571] i2c_hid i2c-ELAN1000:00: error in i2c_hid_init_report size:131 / ret_size:7
[    2.183241] hid-generic 0018:04F3:0401.0002: input,hidraw1: <UNKNOWN> HID v1.00 Mouse [ELAN1000:00 04F3:0401] on 
gauravdhiman@gd-asus-501:~$ xinput 
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ USBest Technology SiS HID Touch Controller	id=10	[slave  pointer  (2)]
⎜   ↳ ELAN1000:00 04F3:0401                   	id=12	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ USB2.0 HD UVC WebCam                    	id=11	[slave  keyboard (3)]
    ↳ Asus WMI hotkeys                        	id=13	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=14	[slave  keyboard (3)]
```

gauravdhiman, 2017-11-04:
Just for your info, I am using Ubuntu 16.04 with kernel 4.2.0.16

gauravdhiman, 2017-11-04:
Another point to mention. Multi-touch works fine with kernel `4.4.0-98` without installing any third party driver. Although the multi-touch feel is not smooth, but it works. I can not move to 4.4.0-98 as many applications like chrome, VS code does not open on that, not sure why - anyway that is a separate issue.

vlasenko, 2017-11-04:
@gauravdhiman This driver is not for your hardware. It is for `FTE100*` Asus TouchPad, not for `ELAN*`