larixer/hid-asus-dkms #50: I/O errors with ELAN1203

spruit, 2017-04-28:
When I install the driver it gives the following output:
_spruit@spruit-G752VM:~/hid-asus-dkms$ ./dkms-add.sh
[sudo] password for spruit: 

-------- Uninstall Beginning --------
Module:  asus
Version: 1.0
Kernel:  4.10.0-20-lowlatency (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

hid-asus.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.10.0-20-lowlatency/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

i2c-hid.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.10.0-20-lowlatency/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

Backing up initrd.img-4.10.0-20-lowlatency to /boot/initrd.img-4.10.0-20-lowlatency.old-dkms
Making new initrd.img-4.10.0-20-lowlatency
(If next boot fails, revert to initrd.img-4.10.0-20-lowlatency.old-dkms image)
update-initramfs.......

DKMS: uninstall completed.

------------------------------
Deleting module version: 1.0
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/asus/1.0/source ->
                 /usr/src/asus-1.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=4.10.0-20-lowlatency -C /lib/modules/4.10.0-20-lowlatency/build M=/var/lib/dkms/asus/1.0/build/src modules...
cleaning build area...

DKMS: build completed.

hid-asus.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.10.0-20-lowlatency/updates/dkms/

i2c-hid.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.10.0-20-lowlatency/updates/dkms/

depmod...

Backing up initrd.img-4.10.0-20-lowlatency to /boot/initrd.img-4.10.0-20-lowlatency.old-dkms
Making new initrd.img-4.10.0-20-lowlatency
(If next boot fails, revert to initrd.img-4.10.0-20-lowlatency.old-dkms image)
update-initramfs.......

DKMS: install completed.
Rebinding to hid-asus
sh: echo: I/O error
sh: echo: I/O error_

See also attached files.

[hid-asus-dkms2.txt](https://github.com/vlasenko/hid-asus-dkms/files/964344/hid-asus-dkms2.txt)
[hid-asus-dkms1.txt](https://github.com/vlasenko/hid-asus-dkms/files/964345/hid-asus-dkms1.txt)

The installation should not be necessary because I am on kernel 4.10, but trying it anyway because the touchpad did nothing at first. Maybe it is because I am using the Ubuntu low latency kernel or because the device is an ELAN1203?

Please help, I love what you are doing for the community!

vlasenko, 2017-04-28:
@spruit This driver is not for your device. It is only for ASUS TouchPads FTE100*. Not for ELAN TouchPads

closed 2017-04-28 by vlasenko

spruit, 2017-04-28:
Too bad, I thought I saw in another thread that this driver is for the ELAN TouchPad. My notebook is also an ASUS. Keep up the good work!