larixer/hid-asus-dkms #53: Installation Succeeds, Touchpad Works, but disappears on Restart

balajeerc, 2017-06-03:
Am able to run the installation and the touchpad starts working immediately.

However, on restart, it appears this module does not get loaded automatically. I have to run `dkms-add.sh` every time.

What might be the problem? 

Asus E402SA-WX227T
Ubuntu 14.04.5
(4.4.0-31-generic #50~14.04.1-Ubuntu SMP Wed Jul 13 01:07:32 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux)

Output from `dkms-add.sh`:

```
~/hid-asus-dkms$ ./dkms-add.sh 
[sudo] password for balajeerc: 

-------- Uninstall Beginning --------
Module:  asus
Version: 1.0
Kernel:  4.4.0-31-generic (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

hid-asus.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.4.0-31-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

i2c-hid.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.4.0-31-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod.....

Backing up initrd.img-4.4.0-31-generic to /boot/initrd.img-4.4.0-31-generic.old-dkms
Making new initrd.img-4.4.0-31-generic
(If next boot fails, revert to initrd.img-4.4.0-31-generic.old-dkms image)
update-initramfs..........

DKMS: uninstall completed.

------------------------------
Deleting module version: 1.0
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/asus/1.0/source ->
                 /usr/src/asus-1.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make KERNELRELEASE=4.4.0-31-generic -C /lib/modules/4.4.0-31-generic/build M=/var/lib/dkms/asus/1.0/build/src modules.....
cleaning build area....

DKMS: build completed.

hid-asus.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.4.0-31-generic/updates/dkms/

i2c-hid.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.4.0-31-generic/updates/dkms/

depmod....

Backing up initrd.img-4.4.0-31-generic to /boot/initrd.img-4.4.0-31-generic.old-dkms
Making new initrd.img-4.4.0-31-generic
(If next boot fails, revert to initrd.img-4.4.0-31-generic.old-dkms image)
update-initramfs..........

DKMS: install completed.
Rebinding 0018:0B05:0101.0001 to hid-asus
```
Output from xinput after installing:

```
~/hid-asus-dkms$ xinput
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Asus TouchPad                           	id=10	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=8	[slave  keyboard (3)]
    ↳ USB2.0 VGA UVC WebCam                   	id=9	[slave  keyboard (3)]
    ↳ Asus WMI hotkeys                        	id=11	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=12	[slave  keyboard (3)]

~/hid-asus-dkms$ xinput list-props "Asus TouchPad"
Device 'Asus TouchPad':
	Device Enabled (138):	1
	Coordinate Transformation Matrix (140):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (260):	1
	Device Accel Constant Deceleration (261):	2.500000
	Device Accel Adaptive Deceleration (262):	1.000000
	Device Accel Velocity Scaling (263):	12.500000
	Synaptics Edges (514):	111, 2683, 94, 1664
	Synaptics Finger (515):	12, 15, 0
	Synaptics Tap Time (516):	180
	Synaptics Tap Move (517):	145
	Synaptics Tap Durations (518):	180, 180, 100
	Synaptics ClickPad (519):	1
	Synaptics Middle Button Timeout (520):	0
	Synaptics Two-Finger Pressure (521):	142
	Synaptics Two-Finger Width (522):	3
	Synaptics Scrolling Distance (523):	66, 66
	Synaptics Edge Scrolling (524):	1, 0, 0
	Synaptics Two-Finger Scrolling (525):	1, 0
	Synaptics Move Speed (526):	1.000000, 1.750000, 0.060588, 0.000000
	Synaptics Off (527):	0
	Synaptics Locked Drags (528):	0
	Synaptics Locked Drags Timeout (529):	5000
	Synaptics Tap Action (530):	2, 3, 0, 0, 1, 3, 0
	Synaptics Click Action (531):	1, 3, 0
	Synaptics Circular Scrolling (532):	0
	Synaptics Circular Scrolling Distance (533):	0.100000
	Synaptics Circular Scrolling Trigger (534):	0
	Synaptics Circular Pad (535):	0
	Synaptics Palm Detection (536):	0
	Synaptics Palm Dimensions (537):	5, 100
	Synaptics Coasting Speed (538):	20.000000, 50.000000
	Synaptics Pressure Motion (539):	15, 80
	Synaptics Pressure Motion Factor (540):	1.000000, 1.000000
	Synaptics Resolution Detect (541):	1
	Synaptics Grab Event Device (542):	1
	Synaptics Gestures (543):	1
	Synaptics Capabilities (544):	1, 0, 0, 1, 1, 1, 1
	Synaptics Pad Resolution (545):	1, 1
	Synaptics Area (546):	0, 0, 0, 0
	Synaptics Soft Button Areas (547):	1397, 0, 1441, 0, 0, 0, 0, 0
	Synaptics Noise Cancellation (548):	16, 16
	Device Product ID (255):	2821, 257
	Device Node (256):	"/dev/input/event7"
```

balajeerc, 2017-06-03:
I am attaching the output of lsmod immediately after install ( `dkms-add.sh` ), and after a reboot.
[after_install.txt](https://github.com/vlasenko/hid-asus-dkms/files/1049871/after_install.txt)
[after_restart.txt](https://github.com/vlasenko/hid-asus-dkms/files/1049872/after_restart.txt)

vlasenko, 2017-06-03:
@balajeerc Thanks for reporting. Please attach dmesg output after restart

balajeerc, 2017-06-04:
@vlasenko Thanks for responding.

Here is the output from dmesg after restart:

[dmesg_after_restart.txt](https://github.com/vlasenko/hid-asus-dkms/files/1050112/dmesg_after_restart.txt)

vlasenko, 2017-06-04:
@balajeerc Looks like the reason for this module to not load on boot is this error message from your dmesg log:
`i2c_hid: module verification failed: signature and/or required key missing - tainting kernel`
This means that kernel module signature verification is enabled on your system. You basically need to figure out yourself what should you do for your Linux distribution to load 3rd party linux kernel modules. 

Or you can try to find Linux kernel >= 4.10 for your Linux distribution, it includes this driver already and it should be signed.

redmcg, 2017-06-05:
@balajeerc I can see hid_asus is being loaded (it is listed in `after_restart.txt`) so I suspect the issue is with your udev rules.

The `dkms-add.sh` tries to add a udev rule at `/etc/udev/rules.d/41-hid-asus.rules`. Can you please confirm it is there?

@vlasenko I believe the `i2c_hid: module verification failed: signature and/or required key missing - tainting kernel` is merely a warning. In case, for example, a tainted kernel is not eligible within your commercial support agreement.

vlasenko, 2018-01-11:
Closing because of inactivity

closed 2018-01-11 by vlasenko